pottedmeat7/OpenVR-WalkInPlace #1: Sometimes Step triggered by HMD x-z movement

pottedmeat7, 2017-11-05:
The threshold for isolating the HMD moving "up and down" seems accurate. However sometimes its also triggered by turning head left and right. Lowering the x and z threshold would make sure the x-z velocity needs to be lower than "up and down". Haven't really found the sweet spot for this but im aware of it, also accuracy of the x,y and z using the Vive Hardware is affected because its not "inside out" it only gives velocity data which is not as accurate as acceleration data. Not sure if this may be just a limit of the hardware...

romulusp, 2017-11-05:
1. reset button is not working
2. value is look better at 0.20-0.25 for all axes (only 10 min of testing, need a lot of work)
speed is to low, maybe double value - always running   or activate speed/ sec
3. i test in onward, Arizona not working, no time now, i will test more games in 2-3 day (someone have to work )
4. maybe  controller movement or together with hmd it will be a better combination. 
Even in this stage i prefer this app more than pocketstrafe ( i'm sure it will be more precise soon)
try to promote yourself (roadtovr, uploadvr, etc), even sell this (people always thinks if something is free it  is not good)
Btw i'm Sharky(my son nickname) on steam
Romulus.

romulusp, 2017-11-05:
onward  - working
gorn - working
arizona - not working (maybe anticheat)
povlov - not working (maybe anticheat)
i try to play  Ui number a little bit, is playable, only real problem is speed
i see you can add arm swing to this app  :) - it will be nice also

pottedmeat7, 2017-11-05:
Did you enable the "Walking" locomotion in Arizona? 
It is working for me, although it does seem slower in that game then in Onward. I think thats due to how they use the touchpad axis for different speeds. 
There have been issues with the UI , no values , values disappearing when typing etc. 
However The reset button I thought actually was working, keep in mind that really atm only the HMD values have an affect, thats why I put a strike-through on the other features, they are currently disabled. I was having major timing issues with the custom Step time value so I think that may be removed for awhile and just be hard-coded.
I have also got a better result with HMD and hand movement because it allows more data to form an accurate step. And this will especially be used for jogging/running, I think I will add a small value for hand tracking for walking, although its a small range to detect...
My current work for jogging/running is promising, I will probably have an update by next week.

Thanks for the game list, i'll add

mmorselli, 2017-11-06:
I read that you don't have trackers, but would it be difficult for you to add trackers' data to HMD+hand data? I'll gladly test it for you. 

The relative position of the two feet can easily (I think) obtain all four directions of movement.

pottedmeat7, 2017-11-14:
I think an additional sensor data either trackers on the feet or something more accurate could help the accuracy issues. The benefit of the trackers is the more broad usage and support of them...
However I think the trackers may still have similar latency issues the HTC vive hardware is built with external tracking and isn't as accurate then inside out tracking. If an additional piece of hardware is needed, it might as well be an inside out tracking device, which would also likely be cheaper then the Vive trackers....

mmorselli, 2017-11-14:
Vive Trackers is accurate as Vive Controller and with exactly the same latency. How would you like to track the feet with an inside-out? I think trackers are perfect for body traking

pottedmeat7, 2017-11-14:
I think if designed specifically for this purpose a switch on the bottom of the foot and an WIFI accessible accelerometer would work very well for this specific purpose. However the Vive trackers provide other uses which is probably why they aren't designed with inside out, the tracking is made external to reduce reliance on the wireless devices and keep everything tracking outside in.

pottedmeat7, 2017-11-29:
Heres my kickstarter to fund the necessary hardware to make this more accurate. The kinect is mostly for a backup solution for people that already have a kinect and don't want to buy the trackers. https://www.kickstarter.com/projects/121765932/openvr-walkinplace

pottedmeat7, 2018-01-05:
One thing I was recently trying out was the "arm swinging" style in FO4VR 
I set "walk touch" to 0 and "Hand Jog Vel" to 0.35 and "Jog touch" to 0.8
with no "button for movement" and then it didnt register simple head knods when I was looking around etc. 
It for the most part would only move when bobbing and swinging the arms (slightly) basically walking with swinging the arms. It was pretty nice.
For now thats as accurate as it will get.
Unless I get started working on the Vive tracker update soon, I think I will be working on some improvements to the algorithm. 
I will put an option to try out a beta algorithm so as to preserve the current functions without risk.

aviglakis, 2018-02-11:
I played around with creating a step detection system. I was using unreal but the logic is probably still valid. Rather than translating direct z movement I: had the software 1. look for a Z change of direction. 2. return the how far the hmd had moved during the last frame before the change of direction had occurred 3. How far the hmd moved during the entire up/down stroke 4. How much time had elapsed during the stroke. 5. How much x, y movement had occured. Being able to fine tune these characteristics left me with a 98/100 step detection with no interference from squatting or whipping the head around, also detected very light steps and heavy steps during this test. It also had an accompanying “intensity” for each step which could be used to fine tune how far a step would take you. Hit me up if you are interested in talking more about any specifics aviglakis@gmail.com

pottedmeat7, 2018-02-19:
Yeah i'm sure there are some improvements that could be made. After testing with trackers it pretty much solves this problem. I could get real involved with the algorithm but I think it feels real good now and can be configured well to work with different setups. 
A real limitation for this driver is that it can only use the speed of movement that is pre-programmed by the game's touch-pad controls. Which is where I would put more focus if I could create more variation of movement speeds, but the touch-pad limits that so there's nothing that I can do.

closed 2018-04-10 by pottedmeat7