pottedmeat7/OpenVR-WalkInPlace #3: Debug information

william-r-s, 2017-12-09:
Some debug tools could help in diagnosing when it isn't working (haven't been able to get it to work myself), some ideas:
Indicator on the overlay when a step is detected
Output on the overlay of recent detected inputs corresponding to the thresholds
Ability to manually trigger walk input to a game (ie. by remapping a controller button), to check whether the input pathway is broken.
Simple steamvr environment that is known to be working with the walk in place input method, to test threshold levels.

pottedmeat7, 2017-12-11:
I am thinking about providing some debug info in the UI, but its quite involved. 
If its not working for you, I would try to uninstall and reinstall the latest version. Ensure first that the UI is launching, then you can check out the logs to see if there is any errors. There is a UI log in your home directory "AppData\Roaming\pottedmeat7\" and there is a driver log in "C:\Program Files(x86)\Steam\steamapps\common\SteamVR\drivers\walkinplace\bin\"

If you have other drivers running I would disable those just to check if they are causing an issue.
I am running on SteamVR beta and also it was working on non beta, so didn't seem to have an issue.
If by "not working" you mean that its not taking steps every time then I recommend taking light "walking" speed steps and find the right sensitivity to make it handle faster steps. Dont just try to swing your arms faster and Head bob higher, because there is an upper threshold on a "step" to make it ignore too quick of movements.

Also make sure the game your playing allows touchpad locomotion. I recommend finding a comfortable value that feel right in the game Compound, because its quick and easy to test.

william-r-s, 2017-12-13:
Tried again, on a fresh install of windows with no extra 3rd party drivers. Wasn't able to get any movement working in Compound.

One question: does the game need to be restarted every time settings are changed?

Found the logs in C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\00vrwalkinplace\bin\win64
[INFO ] 2017-12-12 19:55:01: VRWalkInPlace dll loaded...
[INFO ] 2017-12-12 19:55:01: deviceAddedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: poseUpatedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: buttonPressedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: buttonUnpressedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: buttonTouchedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: buttonUntouchedDetour hook is enabled
[INFO ] 2017-12-12 19:55:01: axisUpdatedDetour hook is enabled
[INFO ] 2017-12-12 19:55:02: deviceActivateDetour hook is enabled
[INFO ] 2017-12-12 19:55:02: Detour::deviceActivateDetourFunc: sucessfully added to trackedDeviceInfos
[INFO ] 2017-12-12 19:55:04: New client connected: endpoint "driver_vrwalkinplace.client_queue.1514693311", cliendId 1
[INFO ] 2017-12-12 19:55:30: deviceActivateDetour hook is enabled
[INFO ] 2017-12-12 19:55:30: Detour::deviceActivateDetourFunc: sucessfully added to trackedDeviceInfos
[INFO ] 2017-12-12 19:55:34: Detour::deviceActivateDetourFunc: sucessfully added to trackedDeviceInfos
[INFO ] 2017-12-12 19:55:58: Detour::deviceActivateDetourFunc: sucessfully added to trackedDeviceInfos
[INFO ] 2017-12-12 19:55:59: Detour::deviceActivateDetourFunc: sucessfully added to trackedDeviceInfos
[INFO ] 2017-12-12 20:05:28: Client disconnected: clientId 1

UI Log
[INFO ] 2017-12-12 19:55:03: Application started
[INFO ] 2017-12-12 19:55:03: Log Config: C:\Program Files\OpenVR-WalkInPlace\logging.conf
[INFO ] 2017-12-12 19:55:03: Log File: C:/Users/William/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/VRWalkInPlace.log
[INFO ] 2017-12-12 19:55:03: Sound effects disabled.
[INFO ] 2017-12-12 19:55:03: Settings File: C:/Users/William/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace.ini
[INFO ] 2017-12-12 19:55:03: VR Runtime Path: C:/Program Files (x86)/Steam/steamapps/common/SteamVR/
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hand threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hand threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd threshold: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd pitch: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hmd pitch: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting walk touch: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting jog touch: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting run touch: No active connection.
[ERROR] 2017-12-12 19:55:04: Exception caught while setting hand threshold: No active connection.
[INFO ] 2017-12-12 19:58:39: Found device: id 3, class 2, serial LHR-FD475D47
[INFO ] 2017-12-12 19:58:39: Found device: id 4, class 2, serial LHR-FFCF5944
[WARN ] 2017-12-12 20:04:07: file:///C:/Program Files/OpenVR-WalkInPlace/res/qml/StepDetectConfBox4.qml:123: ReferenceError: value is not defined (file:///C:/Program Files/OpenVR-WalkInPlace/res/qml/StepDetectConfBox4.qml:123)
[INFO ] 2017-12-12 20:05:28: Received quit request.

pottedmeat7, 2017-12-13:
Hi william-r-s , please try re-downloading the current release and re-install. 
The game does not need to be restarted after changing the settings. 
In compound try using the "both controllers" option at first then if that works you can isolate whichever controller is the correct one.
Because of new issues with in-consistent movement in Fallout VR I will be spending some time developing debug output in the overlay. It seems more of a problem with the way Fallout uses the touch-pad however the values are going to have to be very accurate for it to feel good, so a debug screen is the best way to find the correct values.

closed 2017-12-28 by pottedmeat7