pottedmeat7/OpenVR-WalkInPlace #5: No steps taken for FO4 or Compound. Graph works.

zaneFlux, 2018-01-01:
I'm attempting to get this to work with FO4VR, I've copied your settings from the screenshot on the read me and I've tried controller 1 and 2 and no movement occurs. If I start the graph out seems to register steps but they are not translated into movement. I tried the Compound demo and it was the same, no steps.

mmorselli, 2018-01-02:
Why not set a sort of auto-training module? Every system could be different for some reason, so you could make user walk for a little, jog for a little, make average, min, max, and save a personal profile based on this data

pottedmeat7, 2018-01-02:
auto-training could complicate things if not certain of the setup. The setups _should_ be the same. @zaneFlux can you confirm your not running any other drivers, InputEmulator or other overlays? Did you enable "Direct Movement" in FO4VR ?

mmorselli, 2018-01-03:
in a real environment the setup will hardly be the same, calibration is often present in input systems such as these

zaneFlux, 2018-01-03:
I've got advanced settings, but nothing else is listed is ticked aside from walk in place:
[image]

Is there anything else that could stop it from working?

pottedmeat7, 2018-01-03:
@mmorselli the setup might not be the same but X,Y,Z axis should be the same, as well as how this driver acts. Isolating this issue, if the yellow bars do show in the graph, then steps are being detected, just that controller input is not being applied. This step happens only due to some logic and if its not working it must have something to do with `TrackedDeviceAxisUpdated(deviceId, 0, axisState);` not being run on this system, why, I dont know, OpenVR docs aren't very helpful. 
I've been thinking about focusing my time on fine tuning the algorithm to improve the step accuracy as well as enhance speed / acceleration integration. An auto-configurating the system would be nice but I don't think its the X,Y,Z that's causing this issue anyways.

@zaneFlux Can you send me the logs from "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\00vrwalkinplace\bin\win64"
Also just to verify, you are using Vive with two Vive controllers and Vive HMD, Vive lighthouse with up to date firmware and SteamVR beta?
Also with "WIP" enabled does direct movement work if just using the touchpad in Fallout?

pottedmeat7, 2018-01-03:
@zaneFlux  actually let me draft another release with the logging in there, then I can check out the logs

pottedmeat7, 2018-01-04:
@zaneFlux Please re-install the latest release, then send me the logs from "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\00vrwalkinplace\bin\win64"

zaneFlux, 2018-01-04:
The graph is similar although the Y axis wavelength is shorter.
[driver_vrwalkinplace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1603144/driver_vrwalkinplace.log)

pottedmeat7, 2018-01-04:
The driver is not running at all. Which means its not updating the poses into the system to even check the steps. I would think this is due to some SteamVR version that isnt allowing the driver to run, or firmware. OpenVR documentation is not very good or up to date, and I figure they could update something at any time. I am running SteamVR beta with I think up to date Lighthouse firmware.

zaneFlux, 2018-01-04:
I have steam VR beta and everything up to date:

[image]

I have installed Pocketstrafe but it isn't running when i run WIP, however it installs vJoy (Which also isn't running) could that be something to do with it?

zaneFlux, 2018-01-04:
Didnt mean to close this sorry!

pottedmeat7, 2018-01-05:
Ok I updated OpenVR libraries, that might be the problem.... Try the new release

pottedmeat7, 2018-01-05:
Please refer to this issue https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/7 reverting to non SteamVR Beta works, will keep this other issue open, closing this one.

closed 2018-01-05 by pottedmeat7

pottedmeat7, 2018-01-16:
Hi @zaneFlux try out the new version it has support for the SteamVR updates. https://github.com/pottedmeat7/OpenVR-WalkInPlace/releases