pottedmeat7/OpenVR-WalkInPlace #7: Unable to detect steps in graph

william-r-s, 2018-01-04:
I've installed the new release, and I see activity recorded in the graph but I'm unable to see any detected steps no matter how I tweak the thresholds (and don't see any steps in Compound). 
I've tried uninstalling and reinstalling, and I've tried disabling other overlays (using OpenVRAdvancedSettings and TurnSignal). I'm using the steamvr beta on windows 10.

[VRWalkInPlace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1602329/VRWalkInPlace.log)

pottedmeat7, 2018-01-04:
Can you send me the logs from "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\00vrwalkinplace\bin\win64"

william-r-s, 2018-01-04:
[driver_vrwalkinplace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1603698/driver_vrwalkinplace.log)
The error log is empty

pottedmeat7, 2018-01-04:
The driver is not updating at all. Which means its not updating the poses into the system to even check the steps. I would think this is due to some SteamVR version that isnt allowing the driver to run, or firmware. OpenVR documentation is not very good or up to date, and I figure they could update something at any time. I am running SteamVR beta with I think up to date Lighthouse firmware. I am also on SteamVR beta with windows 10 and it works fine from a clean install...

william-r-s, 2018-01-04:
I reverted to the non-beta steamvr, and it worked! I then switched back to the steamvr beta and it failed. So the current steamvr beta (+ some other factor?) seems to be breaking it.

pottedmeat7, 2018-01-05:
Ok I updated OpenVR libraries, that might be the problem.... Try the new release

william-r-s, 2018-01-05:
Tried new release, works in non-beta but still fails in the same way in steamvr beta (looks like the latest version of openvr input emulator also has a problem with the steamvr beta?)
[driver_vrwalkinplace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1605487/driver_vrwalkinplace.log)
[VRWalkInPlace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1605488/VRWalkInPlace.log)

william-r-s, 2018-01-15:
It stopped working for me again for the non-beta. It looks like OpenVRInputEmulator had to update for the new steam versions, probably the place to look for what changed: https://www.reddit.com/r/Vive/comments/7qea60/release_openvr_input_emulator_v13/

pottedmeat7, 2018-01-16:
Yeah I saw matzman666 had the fixes which will also solve the issues with this driver! Added a new release with the updates, let me know if it fixes it.

william-r-s, 2018-01-17:
Tried it in the current steamvr beta, graph showed steps but they weren't transmitted to Compound or any other game I tried.

[driver_vrwalkinplace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/1637702/driver_vrwalkinplace.log)

pottedmeat7, 2018-01-17:
Now it looks like there is an issue on my system too, perhaps SteamVR didn't update until now... 
I'll have to look into why the exact same code was working Monday and not today...
I probably have a new build soon.
Also remove the entire OpenVr-WalkInPlace folder before reinstalling.

pottedmeat7, 2018-01-19:
Can you try it with just the HMD and two Vive wands. Disable the other 2 trackers. 
I have it working on SteamVR beta and non-beta, getting my trackers soon so probably will be able to test that scenario after that.

william-r-s, 2018-01-19:
I wasn't running it with any vive trackers, just the HMD and the vive wands. (Did other devices show up in the logs or something?)

pottedmeat7, 2018-01-19:
Yes there where two other devices in the logs, thought they were trackers they were actually "tracking reference" devices, could just be the lighthouses so maybe I just never noticed those before. I'll have to check my logs again when I have access to them.

pottedmeat7, 2018-01-29:
Hi @william-r-s I dont know if you normally run OpenVR-InputEmulator also. If you do is that working for you? If you don't can you try installing it from here https://github.com/matzman666/OpenVR-InputEmulator/releases 
Let me know if you can do things like set offsets to your controllers.
Also please run "C:\Program Files\OpenVR-WalkInPlace\Uninstall.exe" remove "C:\Program Files\OpenVR-WalkInPlace" then also remove the 00vrwalkinplace folder from SteamVR drivers path (where you got the log files above) 
Then try installing the new release but use the OpenVR-WalkInPlace-OVRIE-driver.exe installer which uses OpenVR-InputEmulators driver system.

closed 2018-01-30 by pottedmeat7