pottedmeat7/OpenVR-WalkInPlace #8: Same as previous- No steps taken for FO4 or Gorn. Graph works.

LonginusHeart, 2018-02-15:
I've tried Gorn as well and the graph works but I'm not getting any movement. Am I installing incorrectly? Should I be doing anything other than running your installer?

I'm running the OpenVR emulation for Fo4 so I've got your OVRIE installer. I'm also using your screenshotted settings for everything. The trackpad emulation works just fine. Steam VR isn't in beta. 

Thanks for your help on this

pottedmeat7, 2018-02-15:
Sorry, it seems as though there is some issue with the driver callbacks. It works on some setups..., there will probably be more updates to SteamVR and more documentation to allow other ways to callback to devices.

LonginusHeart, 2018-02-15:
Totally understandable. Thank you for the quick response. You seem like a good person

pottedmeat7, 2018-04-11:
Is this still an issue? Does the new release work?

closed 2018-04-19 by pottedmeat7