pottedmeat7/OpenVR-WalkInPlace #9: Vive Trackers Version

mmorselli, 2018-02-25:
It works! For the first time on my system, so it is a big step forward.

But I need to bobble the headset in order to advance, setting HMD Y to the lowest value helps a bit, but IMHO If trackers are active, all other inputs should be disabled.

In addition, a sensitivity setting for trackers would also be useful for use with treadmills .

marcob2178, 2018-03-02:
Hi there. Can u please guide me through the installation and use? I have the vive headset and of course vive trackers (I have 7 trackers). I would like to use the driver to walk in place to different games like for example fallout 4 vr.

Thanks!

CorreAktor, 2018-03-02:
+1 on disabling HMD as an option for movement if you are using the trackers on your feet. There is no need to have secondary validation since trackers can be isolated.

Also, with having to bobble the headset, I usually don't get sick but after just a few moments, I just want to throw up and it doesn't feel natural at all.

Most games don't support trackers, so there isn't the need to have validation of something else, which is understandable if using the wands, which you have the HMD to validate the motion. For the trackers, simply lifting your feet should be enough for almost all games, with a few exceptions that actually use the trackers.

marcob2178, 2018-03-03:
Hi guys, I installed the .exe in steamvr beta, but no overlay appears when starting steamvr. How do I make the program work?
Please help!

closed 2018-04-10 by pottedmeat7