pottedmeat7/OpenVR-WalkInPlace #10: I canv't see the overlay window

marcob2178, 2018-03-04:
Hi. Could you please help me?

I installed your .exe , but when I open steamvr nothing happens. How do I start your program?

Thanks!

pottedmeat7, 2018-03-04:
Can you try to run "C:/Program Files/OpenVR-WalkInPlace/startdesktopmode.bat" does that work? Also did the installer give you any errors or warnings? Do you use OpenVR-InputEmulator also?

marcob2178, 2018-03-04:
Hi there! Thank you so much for your reply!
I did not get any error message when uninstalled your program. It seems I have not installed the openvr-inputemulator. I might have in the past but it is not installed anymore. Should I install it? What gives me more? I have a pair of captoglove that I was trying to use as controllers in vr using some openvr driver with leap motion we sensor. But again, i think I disinstalled the openvr driver because when I installed your program and i tried to use your .exe that is meant to use in case we have openvr driver installed, I was getting a message to use your oth3r .exe that is not meant for openvr installed.

I tried to click on the .bat file but I am getting the message box:

"Another instance is already running "

When the overlay should appear?

marcob2178, 2018-03-04:
I made a typo. I wrote uninstalled , but I meant installed

marcob2178, 2018-03-04:
By the way, I have 7 vive trackers 1.0 version and of course I would love to use your driver for walkinginplace to walk long distances in vr games and in non vr games that we play in vr using vorpx.

marcob2178, 2018-03-04:
It actually now works by clicking on startdesktopmode.bat ! No idea why I had issues . Sorry!

Do u have a video tutorial for setting up the program with vive tracker for waking in place in games?

pottedmeat7, 2018-03-05:
No sorry no video tutorial, every time I try to record  the OVRWIP overlay and Fallout VR crashes or freezes. It should be pretty straight forward.

marcob2178, 2018-03-05:
Do i need to install openvr input emulator along with vr walkinplace drivers, to be able to use the trackers to walk in fo4vr?

I run the .bat of your program and i see the overlay and then I see the graph and the graph is plotting when I moving the trackers. But when I open fo4vr, and I move the trackers , I am not moving in the game :(

marcob2178, 2018-03-05:
There is a user here that have this to say when trying to use walkinplace drivers in fo4vr:

"I have not been able to get a usable profile for WIP using the trackers. Everything that works to move you forward requires a very unnatural walking style, speciflcally the head has to bob a certain way while you move your legs. When using just your arms and head without the trackers it isn't so bad.

Hopefully someone can figure out how to get WIP to disregard the head movement when using the trackers on your feet, as it shouldn't be necessary to use anything other than your feet to figure out if you intended to make a step."

http://steamcommunity.com/app/611660/discussions/0/1618346571418594626/?ctp=2

CorreAktor, 2018-03-05:
marcob2178, that is my comment on steam. I can see the overlay and I can get movement in FO4VR, just not reliably and in a natural manor.

To be clear, the driver and overlay works for me. In the overlay, on the graph page, I can get the yellow grid indicating WIP detects walking AND I can get FO4VR to walk with WIP. What I meant by "not" "usable" is that it isn't consistently detecting steps and that makes it unusable.

What is at issue is that when using the trackers on your feet (I only use two for WIP), you have to nod/bob your head in such a way that it is unnatural and is not consistent. That was the single hardest issue for me to figure out how to make the yellow grid. Since the trackers aren't used in most games, at least two of us are suggesting that when the trackers are used for feet, that HMD and wand detection is not necessary and there should be an option to disable the HMD and wand from being used in the calculation of when a step is made.

If you are seeing the YELLOW grid but no movement in FO4VR, your issue isn't the same.

pottedmeat7, 2018-03-05:
I think I released with a very high Tracker Y threshold. I have been using my profile but I changed my to 0.07. 
The head Bob that is needed doesn't change with or without trackers. You shouldn't need to bob your head anymore then natural walking would require.
I tested this by trying not to move my head while walking (which felt even more unnatural) with a low hmd Y of about 0.1 and it was enough where I would still take consistent steps while trying not to move head. 
So if it's not taking consistent steps it's more likely the Tracker Y setting, try it at about 0.07

mmorselli, 2018-03-05:
likely not all systems are equal

in my system 

- if I move my feet strongly or slowly and do not move my head, I do not walk away

- moving headset strongly or slowly without moving feet it does nothing

- Moving feet just a little and headset strongly, it walks

So, this is no threshold problem. If it were possible, I would like to rely only on feet movement, no headset at all, thank you.

marcob2178, 2018-03-05:
CorreAktor , hopefully pottedmeat7 would make an adjustment so that we could rely only on the feet movements and disable the headset movements :)

It seems that also mmorselli is reporting that he would prefer that option :)

marcob2178, 2018-03-05:
Mmorselli and pottedmeat7 , as for now , I can't even figure out how to make it work to the point were u are right now :(

CorreAktor, 2018-03-05:
And maybe it is that HMD, trackers, AND wand(s) have to move? Again, to me there should be an option to just trigger on trackers if used for feet.

pottedmeat7, 2018-03-05:
It should just be HMD and trackers. I do not experience what you mentioned above. I keep my HMD Y at 0.17 and I barely have to move my head. 
The only other possibility is you have more XZ movement then you are having Y movement (Y has to be greater then the XZ movement), you step differently perhaps... (Maybe this is a treadmill issue)

CorreAktor, 2018-03-05:
I don't have a treadmill, so if it is an issue of XZ (and I'm not the only one having this issue with the HMD) maybe the option to only use the trackers would be a good solution and possibly just the Y axis for the trackers if the XZ is greater than Y? With trackers on feet, there shouldn't be any type of guessing as to what makes a step, just how high.

As far as XZ, my trackers are on the top of shoes, so the foot has a natural tilt when lifting, so yes, the XZ will change while walking. If you also check that if XZ is greater than Y, which disregards the XZ setting to disregard, maybe don't?

mmorselli, 2018-03-05:
@pottedmeat7, could you add a checkbox to totally exclude HDM and consider only trackers? Why make things more complicated than what is needed if they could cause problems?  

after months that the program was unusable on my system, it's finally near perfect thanks to trackers support, if I just didn't have to bob my head too.

marcob2178, 2018-03-05:
@Mmorselli, couod you please share your setting numbers and what you activated ?
Should I check "use trackers?"

mmorselli, 2018-03-05:
[image]

CorreAktor, 2018-03-06:
The only difference with mine from mmorselli is that I have the Step HMD XZ and Tracker XZ both set to 3.0 and I had to use 2nd Controller instead of 1st Controller.

I have managed to be a little more consistent with walking with the trackers, however, there should still be an option to disregard the HMD. I think if the HMD could be disregarded, the step time could actually be decreased and feel more realistic. I feel that when I don't move, I have to start moving my head and once I start moving my head, it then starts over walking things.

marcob2178, 2018-03-06:
Thank you so much!
I set up the setting as above but still I am not able to move in the game when stepping in place. How do we activate the walking/having on the graph the yellow line when walking in place with the trackers?

Sorry to be a pain :(

CorreAktor, 2018-03-06:
1) Check that you are getting the yellow grid in "Show Step Graph" in WIP. If you can't get it to be consistent, you might not be able to replicate in FO4VR.
2) In F04VR, go into settings and find "Direct Movement" and make sure it is set to "ON"
3) Test that your touchpad moves you in FO4VR
4) If the touchpad moves you, try walking like you did to get the yellow grid in step 1
5) If you still don't move, go back into WIP and change "1st Controller" to "2nd Controller"
6) If you still don't move, double check "touchpad no click"  or "touchpad with click" in WIP
7) repeat #5
8) If it still doesn't work, try another basic game to test
9) If it still doesn't work, go outside and enjoy nature, as the Sun God RA is apparently mad at you and doesn't want you inside! ;)

marcob2178, 2018-03-06:
@CorreAktor

I love answer 9) 

That is probably my problem :-D !

I am now be able to move but the movement is not fluid. I mean that It does a few steps in the game and then it stops moving.
It seems that the yellow grid/line is not consistent.  It goes on and off based on certain movements of the head that seems activating the movement.

How do I activate the movement in the game and how do I make the walking steady/continuous? It is not just by walking in place. What head movement activates the walking in place?

CorreAktor, 2018-03-06:
> What head movement activates the walking in place?

That's the issue we all seem to be having. I seem to be able to slightly nod my head back and forth and raise my body slightly on each step. I had to try different ways to hold, nod, and raise my head until I figured out what head movement had to be done along with the tracker steps. It still isn't consistent, however, and is why I think that there should be an option to disable having the head needed when using trackers on your feet.

Glad you made some progress!

pottedmeat7, 2018-03-07:
Did you try to lower your tracker Y to 0.07 ?
I can add a checkbox to ignore the HMD but a natural step but I do think want to catch things like just raising a foot as a step I'm not sure how to do that without the HMD looking forward with an up and down pattern.

mmorselli, 2018-03-07:
in most games there would be no reason to raising a foot not for walking, without trackers feet doesn't exist in VR

After walking I would like to experiment other solutions, such as cycling, to get something similar to virzoom

CorreAktor, 2018-03-07:
I was thinking, since the trackers aren't used for most games, you could also do something like create a command that if you tilt the right tracker 90 degrees (kick your foot back), the system takes that as a command that states that WIP is active. If you tilt the left tracker 90 degrees (kick your foot back), the system takes that as a command that states that WIP is not active. Trying to think of something creative to address the issue you raised, as a possible option if simply removing the HMD causes more problems than it fixes.

marcob2178, 2018-03-07:
Hi @CorreAktor ! Yes, thanks to you guys feeedback I made some progress. Now I see that like you said the issue we are facing is kind of common to all of us.

I like your idea of just having the tracker (walking in place) active  by just lifting the feet vertically, like marching in place. Great that u have different suggestions with a shut off movement too.

Can we maybe think at an option that would activate the trackers (and hence the walk in place), by just clicking at the same time 2 buttons of the vive controllers?

Also, I do not understand why it being said that the walk in place driver/trackers would not work in all games. Doesn't the open-vr walkinplace driver activate this sort of feature indipendentlly from the games? I am thinking how many non vr games we could play in vr using vorpx and using the walkinplave driver, to actually not be sitting on a chair, but walk in place to move....imagine the game "Alien Isolation"

marcob2178, 2018-03-07:
@pottedmeat7 Thank you so so much for your driver. I did not know u had a funding collection otherwise I would have contributed too. I can still do if u need

marcob2178, 2018-03-07:
@pottedmeat7 Would you be able to add in your driver an option to assign keyboard strokes to vive trackers vertical lifting?

mmorselli, 2018-03-07:
@marcob2178 You should already be able to use keyboard emulation, non VR games should all work. Select it in the top menu instead of touchpad

marcob2178, 2018-03-08:
@mmorselli I have tried with alien isolation in vr, but i am not able to see the graph because the pc screen becomes black when I start the game.
Have u tried it on any game that uses arrows?

mmorselli, 2018-03-08:
@marcob2178 not with the latest release, but has worked in the past, try with WASD and notepad, if it works, it also works with notepad and SteamVR desktop (tested)

marcob2178, 2018-03-08:
@mmorselli sorry for my ignorance but what do u mena when u say to try notepad and streamvr?

How do I use notepad in streamvr?

Also , I am curious, what letter do we type when stepping in place using the vive trackers?

Sorry to be a pain

mmorselli, 2018-03-08:
@marcob2178 any text editor you want via standard Windows desktop, when is foreground app should get the keys fired by OpenVR-WalkInPlace (W in case of WASD ). You can check if it works via Desktop function of SteamVR

marcob2178, 2018-03-09:
[image]

[image]

@mmorselli it is writing the letter but steamvr is freezing with this erorr : "vrserver is not running"

mmorselli, 2018-03-09:
@marcob2178 you are using desktop version of OpenVR Walk in Place, this is intended for debug only, try using the native SteamVR Overlay version

marcob2178, 2018-03-09:
@mmorselli ok. Tx! I will try!

I am still puzzled how do I activate the trackers movement /yellow bar. If I shake up and down fast my headset , it seems it activates the trackers

CorreAktor, 2018-03-10:
In your screenshots, change HMD XZ: to something like 3.0. Change the Tracker Y: to something lower, like 0.17 and might as well change Tracker XZ: to 3.0. Lastly, check Use Trackers?.

I noticed that if I try to limit my feet and head movement to the Y access (straight up and down), it is trickers more often, but I still have to move my head more than normal. I think since I spent years in marching band, that when I walk in place, I march, which is ingrained into me to keep my head perfectly still. This, I think is why my gait isn't triggering reliably and why I would like to see an option to disable the head tracking when the trackers are on my feet.

mmorselli, 2018-03-11:
I haven't spent years in marching band, but my head is perfectly still too, when I walk in place ;-)

Each person is different, so we shouldn't really rely on the head but only on trackers.

pottedmeat7, 2018-03-11:
@marcob2178 I believe that there is some conflict while running both OVRWIP and VORPX, it froze for me as well. For now I don't think it will work. 

Im not exactly sure why the HMD movement is causing issues, it could have been something to do with timing of trackers and HMD movement. But I did add a new release with a checkbox so it can be disabled check out the new release.

mmorselli, 2018-03-12:
@marcob2178 WIP toggle (on/off) would be undoubtedly usefull, but IMHO room scale was a total fail in 99% of cases. Room space is always limited and step by step you will always get the wall. I'm using OpenVR Advanced Settings to put a small sign in the center of the room and I never move too far from it. My final goal with the PottedMeat driver is to use it with a treadmill (ROVR 2), I will try it soon

mmorselli, 2018-03-12:
@pottedmeat7 , not really important, but I think the options should be

[x] Use HMD?
[x] Use Trackers?

It's more logical IMHO

mmorselli, 2018-03-12:
@marcob2178 "Button for Movement" with "Button as Toggle" and "Flip Button" options enabled should disable WIP pressing this button, if I'm not wrong. Have you tried?

CorreAktor, 2018-03-12:
@pottedmeat7 Thank you. I will try it tomorrow sometime.

@marcob2178 Most likely, since the trackers are now triggering much more often, try a lower Step Time (seconds). Also, try increasing Tracker Y: and take higher steps when wanting to walk but shuffle your feet while wanting roomscale. As also mentioned, you can set a Button for Movement and rather the Button is a Toggle or not.

CorreAktor, 2018-03-12:
@pottedmeat7 LOL, I couldn't wait, so just tried. Really awesome now!!! Thank you. It does exactly what I had thought it would do when you added the option to remove the HMD from the calculation. 

@marcob2178 I set my Tracker Y: to 0.50 and now if I do more of a march in place, it triggers perfectly. I changed Step Time (seconds) to 0.10 and Walk Touch: to 0.70. You can play around with those three settings to get to where it moves how far you want to move for each step. These settings make it move about two strides forward per step for me, but if I left the Walk Touch at 0.60 it was closer to a single stride forward. The Step Time (seconds) is how long the walk is active per step and the Walk Touch: is how far up the touch pad for speed (also for Jog and Run).

At my current settings, if I shuffle my walk (not raise my foot too high), it won't trigger the walk. If I lift my foot higher, it triggers consistently.

pottedmeat7, 2018-03-12:
@marcob2178 Playing Alien Isolation probably takes some sort of override into the SteamVR system so whether its VORPX or another mod, its probably a similar issue. Not sure why but it seems to freeze or just not work, maybe something can be added directly into this other mod perhaps.

Also for the Room Scale movement problems, that is essentially what the XZ threshold is for. If you are actually moving forward your XZ movement should be higher then it would be walking in place. Setting tracker XZ to as low as about 0.2 should filter out forward movement as well as still allow you to walk in place. You can use the graph to find a more exact number for your Room Scale movement or your style of Walking in place.
Some combination of a lower tracker XZ and higher tracker Y mentioned above could be a very accurate balance.

mmorselli, 2018-03-13:
In the future a sort of auto configuration could be possible? i.e. press "auto configure" button, walk at constant speed for 2 minutes or more, the program calculates averages and set optimal values. Two step if we want to set optimal value for walk and run

marcob2178, 2018-03-13:
Thank you @mmorselli and @CorreAktor for your suggestions and setting optimization . I will try what you both told me and get back to you!

@mmorselli why do u think to still use the treadmill even though now we have this solution of walking in place thanks to this driver? I was looking into treadmills but prices were crazy for the best ones and the cheaper ones have limitation because they block free movements 

Also, I love your suggestion of auto-optimization 

@pottedmeat7 too bad that your driver seems for now now working with vorpx and the alien 's patch.

I am going to try to email both coders and see if they are interested in finding a solution with you

mmorselli, 2018-03-19:
@marcob2178 , I tried the treadmill, the step is a bit different, less tiring and more like a real step, thanks to the smooth surface and shoes with little ceramic pucks below. Is the price worth? I don't know, walking is a bit better and the problem of loosing the center of the room in solved too, but even a tradmill like this is very expensive (from £ 499 to £ 699 ), could be the foundation for studying some DIY solutions to improve walking and reduce the strain

I made a video

https://youtu.be/tJxEgyav29s

marcob2178, 2018-03-31:
@mmorselli that looks cool and thank you for sharing ! :)

Why wouldn't u just use the treadmill without the trackers and walkinplace driver?

It seems that a new katvr kickstarter campaign will be opened soon . Hopefully it will.be affordable.

I own a pair of captogloves and another cool thing would be their full implementation in steamvr to replace the vive controllers. I known they are working on it and they are also working on captosensors to use for walkinplace /locomotion solution.

mmorselli, 2018-03-31:
@marcob2178 ROVR software is really poor and doesn't interact with most games (it only does keyboard emulation), and trackers is more reliable. 

I think if you combine captogloves with a tracker and with the help of OpenVR Input Emulator you can emulate VIVE controllers for most. It allows redirecting tracker to controller and remap keys. I use it with a wireless direct input gun for PSX, so why not a glove?

marcob2178, 2018-03-31:
@mmorselli I get what you mean now and why u use the treadmill with the trackers and the driver. Does it work for all the games? Can it work also with vorpx?

I have no idea how to use  OpenVR Input Emulator with the captogloves and the vive trackers :-(

Would you be able to help me out with some sort of set up please?

mmorselli, 2018-03-31:
OperVR-WalkInPlace seems to work with most games, but you have to try them out, VorpX games uses keyboard, so it should works

With OpenVR Input Emulator you can redirect any OpenVR controller (ie. a tracker) to any other OpenVR controller (ie. first Vive Wand controller), It's really easy, select it from a dropdown and you're done. You can also adjust all offsets. Buttons is another story, what type of buttons Captoglove can emulate? DirectInput (ie. XBOX 360) or SteamVR (VR controller)? Could emulate keyboard too? 

Input Emulator can remap OpenVR controllers buttons, for other type of buttons I wrote some scripts that uses Input Emulator Command Line, the shorter way is via keyboard emulation, but it's also achievable with joystick emulation. I'll pass you the right information based on this 

BTW, OpenVR-WalkInPlace is a fork of OpenVR Input Emulator, they share the same core

closed 2018-04-13 by pottedmeat7