pottedmeat7/OpenVR-WalkInPlace #12: Trackers and sprint

mmorselli, 2018-03-17:
In Fallout 4, how to sprint when using only the trackers? Any idea?

mmorselli, 2018-03-19:
Could be possible an "Onward style" solution? With arms and weapons lowered you run, with your weapons  uplifted you walk

CorreAktor, 2018-03-19:
@mmorselli it would be awesome if there was a way to run while using trackers only. I would think it could be fairly simple, such as "after two steps (is walking), if time between steps < X seconds, walk, else if time between steps is between Y seconds, jog, else if time between steps is > Z seconds, run".

Right now, if I want to run, I just press the wand controller to simulate this, so it isn't that big of a deal to me. It's almost like having a button within WIP but not really.

I would think arm placement would be very hard to program, since you have to fully understand relationships between HMD, wands, trackers and difference in people's bodies (tall, short, long/short arms, etc.), as well as the natural swing of the arms.

Another possibly way would be to use the arm swing, which was used to determine speed but I think tied to the HMD. Maybe if that part got added back into the tracker only option, but only used for Jog/Run. As in, if the trackers are only used, if the arms swing, treat that as Jog/Run, otherwise use just the trackers for Walk. Make it an option for those that don't want it to trigger that way.

Just my thoughts. I don't know how difficult it would be to make any of those changes so not sure what would be the easiest to do.

pottedmeat7, 2018-04-19:
The original issue "How to run" is solved. So closing this. If a feature request is needed please reopen separately.

closed 2018-04-19 by pottedmeat7