pottedmeat7/OpenVR-WalkInPlace #14: how to use with skyrim vr for arm swinging

rayuki, 2018-04-04:
hi mate i want to use your program to use walkinplace with skyrim vr, but i dont have trackers, i saw there is an option to move with arm swinging but i can't seem to get it to work.

i've enabled it and tried all the options but even though your app says its working when i check thru the overlay, when i go back into skyrim nothing happens.

im using direct movement with skyrim on left touchpad.

any ideas?

pottedmeat7, 2018-04-04:
There seems to be an issue with emulating touchpad input and this game. If you move your hands quickly, to trigger running it should make you move when the virtual "click to sprint" is applied. You can also lower the "Hand run" value to like 0.1 
If that doesn't work you can change the game type to "Keyboard (WASD)" in the drop down menu and that will work for Skyrim, although then it can only move at one speed.

rayuki, 2018-04-04:
ok thanks will give it a try.

lasvideo, 2018-04-05:
I think this could be a wonderful addition to Skyrim VR . I want to use just the walking mode. So I DLed and ran OpenVR-WalkInPlace on PC. I have Skyrim in Direct Movement (trackpad smooth locomotion) and OpenVR-WalkInPlace with WIP checked. Set parameters the same as on OpenVR-WalkInPlace info page that show an example. THERE IS NO WALKING MOVEMENT IN SKYRIM, Played around with various OpenVR-WalkInPlace settings. No luck. Anybody have this working with Skyrim VR for PC?

Wolvatron, 2018-04-05:
I had walking with my trackers working in the Skyrim VR setup/tutorial the
first night I had the game. But I forgot to save my settings and haven't
had it working since. I'll let you know if I get it working tonight, but I
probably won't have sufficient time to replicate my old settings until this
weekend.

pottedmeat7, 2018-04-11:
New release should fix issues with input in Skyrim https://github.com/pottedmeat7/OpenVR-WalkInPlace/releases

lasvideo, 2018-04-11:
I got Walk in Place to work after fiddling around with settings. Works nicely. But I have a Vive and after adjusting the touchpad to half walking speed (with a sprint option) , I prefer it. Easier on my knees.  ;)

closed 2018-04-13 by pottedmeat7