pottedmeat7/OpenVR-WalkInPlace #15: NO WALKING MOVEMENT IN SKYRIM

lasvideo, 2018-04-05:
I think this could be a wonderful addition to Skyrim VR . I want to use just the walking mode. So I DLed and ran OpenVR-WalkInPlace on PC. I have Skyrim in Direct Movement (trackpad smooth locomotion) and OpenVR-WalkInPlace with WIP checked. Set parameters the same as on OpenVR-WalkInPlace info page that show an example. THERE IS NO WALKING MOVEMENT IN SKYRIM, Played around with various OpenVR-WalkInPlace settings. No luck. Anybody have this working with Skyrim VR for PC?

Wolvatron, 2018-04-05:
I had walking with my trackers working in the Skyrim VR setup/tutorial the first night I had the game. But I forgot to save my settings and haven't had it working since. I'll let you know if I get it working tonight, but I probably won't have sufficient time to replicate my old settings until this weekend.

pottedmeat7, 2018-04-05:
@lasvideo For some reason the virtual touch input doesn't seem to work in Skyrim. This means that the Walking and Jogging touch commands don't work. 
If its the same issue the "click to sprint" still does work so if you only use the Running setting it will activate that.
You can do that by lowering "Hand Run" to something low like 0.1 
Or another option is to use the "Keyboard (WASD)" option from the dropdown although then it only moves at a single speed.

lasvideo, 2018-04-05:
I tried the  "Keyboard (WASD)" option and nothing happened. Lets make sure I have all my ducks in a row so we know operator error isnt a factor.

1. Im running Steam Beta
2. My system is the original Vive.
3.  I have WIP checked
4. Click to Sprint is selected
5. Both controllers are selected
6. And the rest are set to default.

I do see LOTS of indicated movement on that great graphic that shows motion over time.

So, what am I doing wrong OR what do I need to do? If you could show me a list of steps to get it working, that would be helpful. The immersion factor in Skyrim will really benefit from this mod.

mmorselli, 2018-04-10:
nice paid overlay app that claims to support Skyrim, just bought but not yet tested, no trackers (or Oculus) support at the moment. 

http://store.steampowered.com/app/798810/Natural_Locomotion/

lasvideo, 2018-04-10:
I bought it. But returned it. Arm swinger. Hurts my shoulders (to much archery in real life). Prefer walking locomotion to arm swinging locomotion. I hope pottedmeat is serious about making his work. Havent seen him here for a while though.

gormanj, 2018-04-10:
We need to support this project by becoming patreons if we hope to get updates. I can't get it working well enough to use in Skyrim. However, I plan on trying with two trackers. If it works, I'll be the first to contribute $. If it doesn't well, maybe....
http://www.patreon.com/pottedmeat7

lasvideo, 2018-04-10:
I wont support anything until I see viable results.

pottedmeat7, 2018-04-10:
I know there is an issue with WASD, so at the moment that's not really an option. 

Skyrim most have used a different system to check for touchpad input because it doesn't seem to trigger like it does in FO4VR. 

Mine has been working but it only seems to recognize sprinting, ie. Click touchpad.  I lowered Hand Run to 0 which will trigger click to Sprint for every step for HMD and tracker steps. 
It actually works quite well as once the character loses stamina it slows down to not sprinting. 

Please try this and let me know if it works.

I thought of another way around the different input on this game though. The only other way to make the touchpad actually function (virtually from steamvr isn't working) would be to create a mod. 
This might be really cool because I could get a very wide degree of different speeds which may make me decide to finally calculate more realistic stride speed and distance.

CorreAktor, 2018-04-10:
I decided to do Patreon to help support this. I would rather have an open source project than the proprietary paid app because I believe that the market is fairly small and if the developer stops supporting, people can keep it alive if it is open source. Also, open source allows others to help debug and make suggestions by looking at the code.

pottedmeat7, 2018-04-10:
Thanks @CorreAktor I really would like to make this very full featured and functionally precise as possible. It does take alot of time and energy on my part so appreciate the motivation to make this better! There are a few technical challenges to overcome especially with making it work well for many games. I may start dedicating a bit more time to ironing it all out.

gormanj, 2018-04-10:
I decided to support this as well. Heck, $1 a month. I am going to try it out with my two trackers. I am assuming it still won't work in Skyrim, but if it does, maybe i will up that amount. I need the exercise.

pottedmeat7, 2018-04-11:
So I found some time to rebuild and test and turns out there was just a slight difference in the touchpad input that Skyrim needed in order for it to correctly register. It should be working now try out the new release here https://github.com/pottedmeat7/OpenVR-WalkInPlace/releases . Please try and let me know

gormanj, 2018-04-11:
I just doubled my pledge amount. I had OpenVR-InputEmulator installed. When I first installed the updated OpenVR-WalkInPlace-OVRIE-driver touchpad input did not work (though WASD worked well). Then I uninstalled OpenVR-InputEmulator, uninstalled OpenVR-WalkInPlace-OVRIE-driver. Then I installed OpenVR-WalkInPlace (the standard driver, not the OVRIE version) and it WORKED VERY WELL (aside from some side to side movement I'd like to remove if possible).

lasvideo, 2018-04-11:
Thats great to hear. I look forward to trying it very soon. Walking through Tamriel will be some great exercise.

Bortronx, 2018-04-11:
Hello. I have WindowsMR and it doesn't work for me. Is this Vive only?

CorreAktor, 2018-04-11:
@Bortronx, in the description for this "The OpenVR driver hooks into the HTC Vive lighthouse driver and tracks movement of the HMD and Vive Controllers. It then applies movement into the virtual envrionment."

This would mean that it requires the Vive controller and/or trackers to work.

lasvideo, 2018-04-12:
I got Walk in Place to work after fiddling around with settings.Works nicely. But I have a Vive and after adjusting the touchpad to half walking speed (with a sprint option) , I prefer it. Easier on my knees. ;)

Bortronx, 2018-04-12:
@CorreAktor Sorry I assumed that it would do the same with any other headset tracking.

pottedmeat7, 2018-04-12:
@CorreAktor @Bortronx  I am not quite sure how different headsets are integrated into SteamVR however I believe they are supposed to be mapped to utilizing the same virtual input methods ie. the touchpad on a Vive in SteamVR is the same as the thumbstick on oculus it needs to map it that way in order for the games in SteamVR to function. I assume WMR is setup similarly. However there could be different realtime velocity values from these headsets that would change how the movement is detected via the HMD and controllers so not exactly positive how accurate, or how similar to vive, it would work.

Bortronx, 2018-04-13:
@pottedmeat7
I looked at the graph and it seems to be taking the head tracking values just fine. The issue is that the controller/keyboard input don't work for some reason. I believe it might be an issue with WMR using windows sticks instead of the trackpad but I don't understand why it wouldn't work with the keyboard.

pottedmeat7, 2018-04-13:
@Bortronx can you make a new issue for the WMR issue.

closed 2018-04-13 by pottedmeat7