pottedmeat7/OpenVR-WalkInPlace #17: Moving when button not pressed (Need help)

funinvegas, 2018-04-11:
Took us a while to get it working, but we got movement in both Vivecraft and Skyrim today.  Awesome tool.

We are still facing some issues, the most serious of which is there are movement inputs happening when not pressing the "Button for movement".  Its smooth when holding the button for movement, and head nodes and arm swings cause jerky movement when the button is not pressed.

Smaller issue:  I also don't know what some of the options are, so we have been careful not to change them.  Vivecraft touch/joy input adjusts move speed based on how far to the edge I touch the controller, but when moving with OpenVR we seem to just have one speed?  Is there a setting for this?  Is it the Walk Touch / Jog Touch / Run Touch?

I'm running Vive (not pro).

pottedmeat7, 2018-04-11:
For the first issue, Thank-you for letting me know of this. I tested and I am having the same problem. I may have changed something slightly I haven't focused on that feature for awhile. I will look into why its allowing movement and have an update by this weekend.

For the second issue yes you should be able to move at three varying speeds. The game must support varying speeds using the touchpad by just touching from the center to the the top that is the values 0-1 of "walk/jog/run touch" (0 is the center of touchpad, 1 is the top). For jog/run it uses the varying amount of arm swinging to switch between the speeds.

pottedmeat7, 2018-04-15:
the bug where "Button for movement" was not completely disabling movement has been fixed please try the new release. 
Also sometimes it is hard to tell if you are walking / jogging because the touchpad input only allows for a small range of speed changes I added data in the graph to show when jogging and running are detected.

closed 2018-04-15 by pottedmeat7