pottedmeat7/OpenVR-WalkInPlace #18: Skyrim VR walking/Jogging randomly STOPS

BrentW89, 2018-04-13:
I did some more testing in Skyrim VR with WIP and various touchpad settings (both with trackers/no HMD as well as with the arms/HMD implementation). In all scenarios, the movement will randomly stop every 5 to 30 seconds of movement. This can happen if I'm walking or jogging (not sure if I've seen it while sprinting). However, it happens so often (usually within seconds) that it makes the driver unusable!

When this happens, I try moving faster, slower, bob head, swing arms more or less, touch the pad, or something until it gets "unstuck" and starts moving again. However, the problem returns quickly. I even try exaggerated arm swings and head bobs to make sure I'm always triggering the threshold, but that doesn't seem to eliminate the issue.

Are you seeing this? If not, what might be causing it? If your install works in Skyrim VR with walk, jog, sprint, then a screenshot of your working Skyrim implementation with and without trackers would be very helpful! I've tried many different settings to no avail.

pottedmeat7, 2018-04-13:
I am not experience this issue. If you used the settings example from the Readme https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/Readme.md
Then it should detect steps and be very consistent. Obviously if you are using trackers you would check "Use Trackers?" and possible "Disable HMD". Are you changing anything else with the settings?

I would recommend not faking the step, ie. it will work best if its as natural as possible. 
Unless there is something different with the way you walk then it should be very consistent.

Try and look at the graph also to see when the yellow bars appear and stop, to see if you can see why it did not detect a step. Could be a threshold is not met or XZ movement is too extreme.

BrentW89, 2018-04-15:
I copied the settings from the Readme.md, turned off HMD and turned on trackers, and it worked great for my entire session!  I even tweaked the walking/jogging speed and it kept working. I saved the profile as "default" and when I came back and played it later it no longer worked (Despite the docs saying this should auto-load, I had to manually re-load the profile for some reason).  Anyway, the movement kept randomly stopping again like before. 

I tried re-entering all the original values from the readme.md again, double checked it it all, but this time it still didn't work. At one point, the movement stopped responding altogether. I tried switching between controller 1 & 2 along with trying lots of other things, but nothing worked. That's when I gave up. The app is definitely very flakey and inconsistent with Skyrim VR where behaviors change at times even when I've not changed any settings or my walking style.

Another weird issue (maybe it's related?): When movement is responding, if I swing my left hand to the left, my character strafes left as if I had touched the trackpad! The right controller does not do this.

Lastly, I've never installed that OpenVR-WalkInPlace-OVRIE-driver.exe as my understanding was it was optional. When I tried it over a week ago, I think it gave some kind of error.

BrentW89, 2018-04-15:
(Also, it would be helpful if you explained what the difference between the 3 graphs are)

pottedmeat7, 2018-04-15:
If it is changing this much between plays I would say it could be other external tracking issue with your playspace perhaps shiny objects or mirrors throwing things a bit off... 
Or you could have some slight different setup, im really not sure. I haven't had any of these issues.

The issue "if I swing my left hand to the left, my character strafes left" in Skyrim direction is controlled by the controller direction and movement is controlled with the left hand. If you swing your left hand to the left pointed it to the left you will move in the direction your controller is pointed. The right controller's direction is not considered for direction in Skyrim.

pottedmeat7, 2018-04-15:
I am noticing an issue that sounds like maybe its the same thing you are. It is only happening when run is triggered. I am also noticing some buginess with just manual click to sprint in Skyrim. Sometimes it wont hold the sprint and it will stop even though i'm physically clicking the button. Also It uses the controller direction to control which direction the "sprint" movement moves in.
Even when the controller is pointed only slightly to the left and I click the touchpad it does kind of like a strafe/side step sprint movement. 
Im noticing that when my WIP system triggers a click to sprint that I have to keep the controller pointed pretty straight in order to not trigger Skyrim's side step movement, and also every now and again it will be stop and go. 
I can reproduce the issues even with the physical click to sprint so don't think its really solvable completely through this driver.

pottedmeat7, 2018-04-15:
Also really important and may seem like a bug. If your controller is past a point pointing more vertically or actually more towards yourself, ie backwards or upside down. 
That will make you move backwards. Like pulling a joystick hard back. 
I've noticed this happen when I swing my arms so much the controller actually points backwards. Which will halt your forward sprinting and come to a complete stop.

pottedmeat7, 2018-04-19:
Not reproducible. No other information provided. Closing.

closed 2018-04-19 by pottedmeat7

MelonBoyy, 2024-05-29:
Had this same issue occur to me, no matter what settings I changed inside the game or overlay. Slow movements with the headset register but if you turn just a bit too quickly even for the smallest second, your character stops moving. I tried making multiple models, only tracking the controllers, only tracking the headset, and yeah it just keeps happening.