pottedmeat7/OpenVR-WalkInPlace #21: Walking Disabled When Turning

czogby, 2018-04-15:
Turning in place (even slightly) causes all steps to not be registered (while still turning). I have no idea why - my XZ thresholds are maxed out.

I've read the readme and tried a lot of things, but nothing is fixing it.

Any idea what's going on?

pottedmeat7, 2018-04-15:
I'm not experiencing this issue. I do experience it if I turn very quickly while walking in place. 
It's possible that the XZ values are exceeding the Y values which would also disable the step.

czogby, 2018-04-15:
Interesting.  Was that by design that XZ values exceeding Y values would disable the step?

Any ideas how I could work around it?  Or would the code need to be changed?  Not being able to turn while jogging is pretty immersion-breaking.

pottedmeat7, 2018-04-15:
Its by design, it will filter out unnatural walking steps. If your XZ is greater then the Y you must be turning pretty quickly. 
Its not happening for me on slow turning. Only if I turning quickly.

czogby, 2018-04-15:
Okay it does look like if I jog with my knees way up high, and if I only rotate extremely slowly, steps aren't disabled.  But that's pretty awkward and would get really tiring, and limits mobility compared to using the trackpad.

Do you think you could add an option to turn off the check for XZ > Y?

pottedmeat7, 2018-04-16:
There really shouldn't be natural step movement that makes XZ > Y I would like to keep the criteria as close to a natural step as possible. 
There must be some other issue with what is happening and dont want a fix that doesnt address the actual issue. I take very natural simple steps and am able to turn naturally and it detects my steps very consistently, so im not experiencing this issue. 
Maybe if you have similar issues to others such as unsupported HMDs (WMR doesnt track correctly for instance) or tracking problems in your play space (shiny objects or something) Im not sure, but my WIP is very natural and easy and consistent.

czogby, 2018-04-19:
Just wanted to make sure you understand what I meant.  With my "minimal jog" motion, the trackers only move maybe 3 inches in the Y direction, at the most.  Now imagine you wanna do a 90 degree turn in a half-second.  Your outer foot is gonna move about 6 inches horizontally as it's moving upwards.  I just tried it out and watched what my feet were doing.

Does that make sense to you?

closed 2018-04-19 by pottedmeat7

czogby, 2018-04-19:
I don't understand.  It IS a realistic movement.  In fact, I would likely want to turn FASTER than that sometimes.

Is it really that much effort to add an extra tick box for anyone who prefers this option?

czogby, 2018-04-19:
Or could you at least point me to where in the code I could change it myself?

czogby, 2018-04-19:
Just one more quick attempt at explaining what I'm talking about: imagine you're playing a console game, like Mario 64 or whatever.  Is Mario able to turn 90 degrees while running in a half second?  Or less?

czogby, 2018-04-19:
Example:
https://youtu.be/XnqGFn0Byg4?t=4m1s

But again, if you don't feel like changing it, can you please point me to where in the code I can change it myself?

pottedmeat7, 2018-04-19:
You have to change the boolean checks in the `upAndDownStepCheck` function in 

https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/client_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp

czogby, 2018-04-21:
Would you have a minute to point me in the right direction for how to compile it?

It doesn't compile in Visual Studio - it gives a bunch of errors about missing QT source files.  I downloaded QT but can't figure out how to use it with this project.

czogby, 2018-04-25:
Any chance you could help with a compiler issue I'm having?

I followed all the steps in the readme, but I always get the following errors:
cannot open file 'libboost_date_time-vc141-mt-gd-1_65_1.lib'
cannot open file 'libboost_regex-vc141-mt-gd-1_65_1.lib'

These errors are immediately preceded by two warnings that seem like they could be the cause, but I can't figure out how to fix them:

TargetPath(C:\Users\Charles\Downloads\OpenVR-WalkInPlace-master\client_overlay\bin\win64\OpenVR-WalkInPlace.exe) does not match the Linker's OutputFile property value (C:\Users\Charles\Downloads\OpenVR-WalkInPlace-master\client_overlay\bin\win64\OpenVR-WalkInPlaceOverlay.exe). This may cause your project to build incorrectly. To correct this, please make sure that $(OutDir), $(TargetName) and $(TargetExt) property values match the value specified in %(Link.OutputFile).

TargetName(OpenVR-WalkInPlace) does not match the Linker's OutputFile property value (OpenVR-WalkInPlaceOverlay). This may cause your project to build incorrectly. To correct this, please make sure that $(OutDir), $(TargetName) and $(TargetExt) property values match the value specified in %(Link.OutputFile).

Have you seen this issue before?  Any ideas?

pottedmeat7, 2018-04-25:
Im pretty sure you can ignore the warnings. I've seen the boost errors before, but forget exactly how I fixed it. You can always build boost from scratch for your specific setup that may help.

czogby, 2018-04-26:
Thanks.  I could try that and see if it helps.  Side question: how much time would it take you to just take my edited version of WalkInPlaceTabController.cpp and compile it for me?

pottedmeat7, 2018-04-26:
Yeah patreon is good. Yeah should be a pretty simple build if there is not much changes.

czogby, 2018-04-27:
Alright, sounds good.  Here's the edited file:
https://www.dropbox.com/s/t72s7qygyk47gs6/WalkInPlaceTabController.cpp

pottedmeat7, 2018-04-28:
Ok how about $15? To my patron page then I can build it, there is some new changes to the UI only for changes for the accuracy buttons so Im leaving that otherwise it will take more time.

czogby, 2018-04-28:
That works.  I just pledged $15 on Patreon.  Though PayPal might have been easier for a one-time donation.

Let me know when it's ready.  Thanks.

pottedmeat7, 2018-04-29:
Ok here is the installer. Also the UI is the updated one enjoy it before its released ;) https://drive.google.com/file/d/1dVEB-Nm9L5cndRJecZSj_rrOO-NAGgQH/view?usp=sharing

czogby, 2018-04-29:
Nice - this works much better for me.  I think I'm gonna keep my trackers after all.  I'll even give you an extra 5 on the pledge just because of how much of an improvement this is.

By the way, one more simple suggestion I thought of: it would be nice to have more options under the menu for which button is activated by steps.  The game "Vertigo" uses the grip button as the "walk forward" button, but that's not an option in OpenVR-WalkInPlace.