pottedmeat7/OpenVR-WalkInPlace #23: Need to jog before walking? (also using vive trackers)

Disastorm, 2018-04-17:
I just installed this the other day (with vive trackers) and was using it with VRChat and the only way I could get it to walk was by first moving my hands to get it to jog and then stop moving my hands and continue moving my feet and it would be walking.
Is this intended? Its kind of weird since if I want to walk forward I have to burst forward with a jog first, and I could see that being a problem in some games.

pottedmeat7, 2018-04-17:
Check out the graph and then you can really see when walk (yellow lines) or jog (orange lines) is triggered. The walk uses just the HMD or Tracker movement to trigger it so your hands dont need to move at all. 
To really feel a difference between walk and jog try lowering the "walk touch" and you can also raise the "Arm swing Jog" so it doesn't trigger jog to easily.

Disastorm, 2018-04-18:
I'll try it again but I'm fairly certain walk was not triggering at all unless I started jogging first.

Disastorm, 2018-04-18:
I tested it out and it actually works in games like Organ Quarter, but its just in VRChat where it has this weird issue.

pottedmeat7, 2018-04-18:
I think for VRChat it only works if "walk touch" is set to 1. Im really not sure, I havent tested in that game for awhile. Im pretty sure there is only one level of speed in VRChat.

Disastorm, 2018-04-19:
there is not only one level of speed, if you use trackpad its based on how far from the center you are touching, also like I mentioned, if I jog first and stop jogging but continue moving my feet, it will indeed be "walking" at my set walk speed in OpenVR-WalkInPlace

pottedmeat7, 2018-04-19:
Is this issue only in VRChat? I dont use VRChat too much so I forget how it works but from what I remember it was a bit different

Disastorm, 2018-04-22:
I only tried it on 2 games, it worked on one and vrchat was the one where it didn't work. If you remember vrchat being weird then maybe its the only game with the issue.

Disastorm, 2018-05-11:
based on my last comment, I wonder if you think it would be possible?

Most games that support trackpad movement generally allow you to move in all directions of the trackpad ( not only forward ).
I guess it might take alot of work to update, but I think in theory, if you could have a way to calibrate the orientation of the foot trackers to have a "forward" direction, you should be able to get the orientation of the hand controller that is using the trackpad movement, and calculate the relative angle that the foot is oriented to allow players to move based on their foot orientation ( so that they can move their hands, aim, etc freely while walking in the same direction the whole time ).

closed 2018-07-06 by pottedmeat7