pottedmeat7/OpenVR-WalkInPlace #24: using the W key for going forward

Jenssons, 2018-04-18:
now i imagine this is a thing you cant do but i have this vr tread mill called rovr what triggers the W key for moving forward, what is awesome for the vr minecraft mod. I would love to be able to use it for fallout 4 vr or some thing else like skyrim so is there a way to allready do that or could this be a possible update?

mmorselli, 2018-04-18:
OpenVR-WalkInPlace already supports WASD keys as output

I own the ROVR too, and I prefer to use trackers instead of microphone

https://youtu.be/tJxEgyav29s

Jenssons, 2018-04-18:
i want to use it as a input not as a output also i use oculus and its kinda pointless to have a rovo with trackers

CorreAktor, 2018-04-18:
@Jenssons maybe you need to try OpenVR Input Emulator to see if you can map it?

As you mentioned, this project is probably not what you need because it is to be able to walk without something like ROVR. To your statement, it's kind of pointless to have ROVR with this app, because the purpose of this app is to use the HMD, hands, and/or Trackers to have movement without needing something like ROVR.

My opinion is that you should really look at Input Emulator and have the group maintaining that work on what you are asking, as it sounds like that is more appropriate than this.

Jenssons, 2018-04-18:
but there arent any input Emulators for this :L

mmorselli, 2018-04-18:
So, what you want is convert "W keypress" into trackpad touch? You can achieve this with AutoHotKey and OpenVR Input Emulator command line. 

in AutoHotKey:

```
w::
Process, Exist, client_commandline.exe
If Not ErrorLevel
    Run trackpad-up-press.bat,,hide
	KeyWait, w
    Run trackpad-unpress.bat,,hide	
return
```

input emulator scripts:

trackpad-up-press.bat

```
CALL config.bat
client_commandline.exe buttonevent touchandhold %ctrlid% 32
client_commandline.exe axisevent %ctrlid% 0 0.0 1.0
client_commandline.exe buttonevent pressandhold %ctrlid% 32
```

trackpad-unpress.bat

```
CALL config.bat
client_commandline.exe buttonevent unpress %ctrlid% 32
client_commandline.exe buttonevent untouch %ctrlid% 32
```

config.bat

`SET ctrlid=3`

Jenssons, 2018-04-18:
where would i put the trackpad-up-press.bat and trackpad-unpress.bat files?

Jenssons, 2018-04-18:
ok i made a rovr.ahk with all the bat files in the same folder ok so after i run the file what do i do then?

mmorselli, 2018-04-18:
You need client_commandline.exe from OpenVR Input Emulator too, in the example above trackpad-up-press.bat trigger a trackpad continuos press on controller ID 3. Make sure you have read AutoHotKey and OpenVR Input Emulator docs.

closed 2018-05-14 by pottedmeat7