pottedmeat7/OpenVR-WalkInPlace #25: Trackers not recognized

BrentW89, 2018-04-19:
I have the WIP and tracker box checked and settings exactly like your screen shot example. However, the trackers aren't recognized (they were recognized for a while, in your previous version I think). The line is flat on the graph screen. However, the trackers are paired and lit green. I even tried pairing them again. Also rebooted. I've also uninstalled your driver, deleted the folder, and reinstalled (it still shows all of my old profiles after the uninstall by the way). This time I tried installing the OpenVR-InputEmulator-v1.3.exe and then the OpenVR-WalkInPlace-OVRIE-driver. Trackers are still not recognized.
 
Note: HMD and hands are recognized.

pottedmeat7, 2018-04-19:
If this is an issue with this driver then its down to the level of OpenVR not recognizing trackers as its documented value `vr::TrackedDeviceClass_GenericTracker` 
So the issues that you are having, that I cannot reproduce, are issues that from my end there is nothing to be fixed. I look for `vr::TrackedDeviceClass_GenericTracker`  and if that isn't working then there is nothing I can do.

closed 2018-04-19 by pottedmeat7

BrentW89, 2018-04-22:
This time I uninstalled walkinplace, rebooted, and reinstalled OpenVR-WalkInPlace.exe. This fixed it and it worked fine for the entire session I played, but then the game crashed with a VR server not running error displayed in the SteamVR desktop app. I rebooted and it was back to the trackers not being recognized. I uninstalled and reinstalled and it fixed it again. Eventually I received another VR server error and the last uninstall/reinstall has not fixed it so far. I may try again later. Maybe I didn't reboot at each step or something.

When I'm playing Skyrim, I can play it for hours without openvr walk in place running and never get the VR Server is not running error in the SteamVR app on the desktop. However, within an hour or so of playing with your app (and sometimes within minutes), it will crash. When it crashes, could it be corrupting a setting that disables the trackers until an uninstall? That's what seems to be happening. If it didn't do that, it would be very helpful. 

Also, you say you can't do anything related to the trackedDeviceClass, but the source code is on Github, right? The description I said warned that updates in steam might break the driver. Could a recent update be causing problems? (I'm running the latest version of Steam and Skyrim, but not the beta. I also have the Vive Pro). There is something with your app or the driver that is unstable. 

Is there a debug log or something we can do to track down what the problem is?

pottedmeat7, 2018-04-22:
There is nothing that is saved with my driver after reboot. There is no reason why a reboot would fix anything or reset anything within this system.
Issues with SteamVR on the other hand do prefer reboots. When you have tracking issues with your equipment or SteamVR version or lighthouse drivers so on etc. 
"Also, you say you can't do anything related to the trackedDeviceClass, but the source code is on Github, right? " My point is that I cannot do anything more simple and precise then what im already doing. In the source I filter by the device class `vr::TrackedDeviceClass_GenericTracker` .  That is what trackers will show up as from SteamVR. That is what my source does. If the trackers aren't showing up with the correct device class then there is nothing I can do thats a SteamVR or other issue. Ie if you trackers aren't showing up then something is not registering the correct device class to my the Steam VR server system, that is not something that I can fix.
There is some changes to the VR server and driver access for OpenVR so its possible that the version you are on wont function correctly, its somewhat out of my hands as I cannot fix issues with SteamVR or OpenVR and I can only use there very limited documentation to make this system functional and for me it works flawlessly.  
As the way you described these issues I can be pretty certain that they are from other issues other then just my system. So I cannot help you fix everything with your steamVR setup.

BrentW89, 2018-04-24:
MORE INFORMATION: 
Yesterday I got WIP to start working again by uninstalling, deleting the WIP program files folder (that fails to delete when I uninstall), and rebooting. It then ran pretty well for hours.

Today it has stopped working again (unsure why), but I noticed that not only does the Trackers not show up on the graph, but the HANDS do not show on the graph! The HMD does show up though. Does this give you a clue as to what might be happening?

Also, I use Vortex to manage mods in Skyrim VR. Is there specific installation instructions for using Vortex or mod managers? Maybe this is breaking WIP?

BrentW89, 2018-04-26:
> If you are using another OpenVR or something that overrides SteamVR's
> virtual detection of devices then that could have an affect on the devices
> showing up or not. OVRWIP does not seem to run with VORPX for instance.

I'm not running Vorpx and I can't think of anything I'm running that is
altering the detection of devices. I did try a steam app called Natural
Locomotion (does not currently support trackers), but that was only after I
couldn't get OVRWIP to work reliably.