pottedmeat7/OpenVR-WalkInPlace #26: Feature suggestion: Option to interpret touchpad touch location to signify direction relative to controller

Blindmikey, 2018-04-22:
A feature suggestion to add an option to allow the touchpad to send direction-intent.

Use case would be selecting the touchpad 'touch' to be the button to enable walk-in-place movement with the touch location signifying intent of direction, so touching the left side of the controller's touchpad and then walking in place would make leftward strafing movement. 

Thoughts?

pottedmeat7, 2018-05-14:
One goal of mine when building this tool is to limit any physical, analog, input as much as possible ie. touching or holding any button. 
Instead I would like to detect the direction the player is facing using other means. I think the controller direction works well for most games. I think it also would work well to use the tracker direction when strapped to your feet facing forward. This is a bit more difficult to figure out, but that may be a future enhancement if I can find the time.

Blindmikey, 2018-05-15:
Tracker enhancement would be ideal for those with trackers. Only use case where pointing a controller doesn't work, is when two hands are required for an action while user wants to strafe ( like using a bow & arrow while strafing ). 

Thanks for this awesome driver, keep up the great work!