pottedmeat7/OpenVR-WalkInPlace #30: Feature Request: Allow step time to below .05

CorreAktor, 2018-05-13:
@pottedmeat7 , on the next build, could you change the step town to allow going down below .05 seconds? I am using trackers only in both F0VR and SkyRim. I have it trigger perfectly for me stepping but what I have noticed is that it will trigger immediately when I make a step, but the delay in stopping isn't almost immediately. It is one of two things most likely. I am assuming that the step time is how long the walk virtual button is pressed when you sense movement. I am also assuming your algorithm tries to figure out when you are stepping using some time interval. If I could reduce the step time below .05 to .03 or even .01, I could see if reducing it will produce an almost immediate stop but not below whatever time interval you determine the step, unless that is a minimum of .05.

I've mentioned this before, but it would also be nice to have the trackers, when only using the trackers, have the algorithm figure out how fast you are stepping to trigger between WALK, JOG, and RUN.

Once again, I have to say that your driver is awesome and that it should be part of the Steam VR Open VR standard for all games.

pottedmeat7, 2018-05-15:
Oh wait scratch that. There is an issue when using the "disable HMD" option the same "step time" variable is not used. The time is mostly used to count time between HMD steps. So I'll have to make it work for either.

CorreAktor, 2018-05-15:
Thanks. So I am guessing that changing it in the ini wouldn't work until you change that. I had found where you put the INI and was going to try and change it to .02 to see if would work but won't try it based on your last post. Hopefully it isn't a major change and you can get it into the next build.

pottedmeat7, 2018-05-19:
Ok try out the new release https://github.com/pottedmeat7/OpenVR-WalkInPlace/releases you should notice a significant different between a value like "0.5" and "0.05" now. The variable time just need to be applied when using "disable HMD".

closed 2018-05-19 by pottedmeat7