pottedmeat7/OpenVR-WalkInPlace #31: headset movement not detected with Windows Mixed Reality devices

godzan, 2018-06-12:
Hi, I am trying to use this with my Windows Mixed Reality devices, but it isn't detecting any movement from them. I can loan you a Windows Mixed Reality Headset for testing if you like. I will pay for shipping if you want to do this.

pottedmeat7, 2018-06-12:
If you loan me the headset I will spend time trying to figure out the issues with the system not working. It probably is something relatively simple, different controls or tracking mapping. So I should be able to figure it out if I have the headset to test.

godzan, 2018-06-12:
I can ship it on Thursday. Do you have a  address you would like me to send it to?

pottedmeat7, 2018-06-12:
Can I have your direct email, I will send you my address.

godzan, 2018-06-12:
evowil4lan@gmail.com

pottedmeat7, 2018-06-25:
Do many games use the WMR touchpads or do they mostly use the thumbsticks?
I notice that Compound uses the touchpad but Skyrim uses the thumbsticks. Im not sure which button the touchpad maps to as the thumbsticks seem to map to the Vive touchpad, there must be another undocumented button type for the WMR touchpads...

godzan, 2018-06-25:
Every game is different. Some use the touch pads and some use the thumb
sticks.

pottedmeat7, 2018-06-29:
Can you try out the latest release and let me know how it works for you? Im still not sure which axis the touchpads are using but the thumbsticks were working in Skyrim, I also was having tracking issues but it was working earlier in the week, I think my WMR setup is affected by some shiny objects in my room...

godzan, 2018-06-30:
I tried the latest release and it works. The movement is very lagy, and
there is adelay between steps. I will have to spend more time adjusting it
unless you have figured out settings that work well.

I also dont see the settings show up for the steam

godzan, 2018-07-01:
https://www-windowscentral-com.cdn.ampproject.org/v/s/www.windowscentral.com/how-enable-windows-mixed-reality-joysticks-any-steamvr-game?amp_js_v=a2&amp_gsa=1&amp&usqp=mq331AQECAE4AQ%3D%3D#referrer=https://www.google.com&amp_tf=From%20%251%24s&ampshare=https%3A%2F%2Fwww.windowscentral.com%2Fhow-enable-windows-mixed-reality-joysticks-any-steamvr-game

Do you think there is a way to force or trick games to use the thumb sticks
instead of the touch pads? A lot of the games that use the touch pads look
like their emulating the vive controller.

godzan, 2018-07-01:
https://www.reddit.com/r/WindowsMR/comments/7jen9e/tutorial_how_to_use_analog_sticks_in_steamvr/

https://steamcommunity.com/app/719950/discussions/0/2906376154312217713/

I also found these. I will try them some time this week.

pottedmeat7, 2018-07-02:
You can redirect input from one type to another. I dont have anything built in OVRWIP to redirect input, but I can emulate either via WIP. I think there is a bug in the driver system that isnt mapping both the touchpad and thumbstick options. So either type of game can be used, will have another release soon. I also was experiencing the delay with steps there was a new bug when I updated jogging/running, this will also be fixed.

pottedmeat7, 2018-07-04:
Ok found the fix for distinction between touchpads and thumbsticks. So now both work! It feels just like my Vive setup now! Try it out and let me know. Thanks again for lending me the WMR HMD! 👍

godzan, 2018-07-04:
The touch pads are working really well in games that use them. The movement
is a lot smoother that it was before. There are a few brief pauses every so
often, but it's nothing I haven't gotten use to. You did a really good job
with this release, and I am shocked at how quickly you were able to get it
out. Good Job!

closed 2018-07-05 by pottedmeat7