pottedmeat7/OpenVR-WalkInPlace #32: jumping

caddilacbob, 2018-06-19:
you haven't given us a way to jump, as an arm swinging api this far exceeds natural locomotion, especially given that this is free however as they don't seem to be compatible i am going have to stick with natural locomotion

pottedmeat7, 2018-06-19:
Yes I have thought about having a jump simulation. However not only does this seem somewhat un-safe but also not really a significant improvement. Maybe I will try to implement it and see if its fun and if its safe so I know where my feet are...

pottedmeat7, 2018-07-23:
I was thinking about jumping and I think really jumping could be done on a real 1-1 scale via the game. It actually already works in room scale, when you jump your character does go higher ;) . I think this would be the best long term functionality for VR Games. But I think jumping just adds another layer of complexity and uncertainty that I cant take on developing for. I find it suitable to push the button and if I feel like it I can also jump in place...

cdibbs, 2019-04-27:
I agree with @caddilacbob. Right now, I walk/run in place (not just arm swinging) and it feels so natural. Props to you for that! The only thing missing is the jumping. That feels like it would complete it, taking this plugin as far as it can go without treadmills and weird/awesome sci fi solutions. :-) Anyway, thanks for your work on this!