pottedmeat7/OpenVR-WalkInPlace #34: Error "Could not find HMD"

Ubranek, 2018-06-24:
Hello!
I just Install "OpenVR-WalkInPlace.exe" and try to run "startdesktopmode.bat" and see only this error - don't catch what can be wrong. 
[image]

The same happens with execution "OpenVR-WalkInPlaceOverlay.exe" and after I have build and ran openvr-inputEmulator itself from sources. 

SteamVR worked fine with HTC Vive, but i want to run games without it - is it possible at all?

pottedmeat7, 2018-06-24:
What folders are listed in `C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\` ?
Usually the only time I get the "Could not find HMD" error is when my HMD or lighthouse is not powered on. 
Are you using an HTC Vive? Check that your not on SteamVR beta by checking the SteamVR software in the Steam client.

Ubranek, 2018-06-24:
SteamVR not beta.
Right now i do not have any real headset or lighthouse and try to find out how i can emulate them,
I guess there is some mistake in .vrsettings (.vrresources and .vrmanifests) files in emulated drivers - do not really understand what they have to be like.
Added "enable" : true in 00vrwalkinplace\resources\settings\default.vrsettings and have new (?) error message

[image]

pottedmeat7, 2018-06-25:
If you are trying to virtually emulate HMD or controllers etc. 
This is probably not the project your looking for... 
I removed all of the functions that allow for virtual emulation of controllers or HMDs that aren't physically there.
OpenVR-InputEmulator still has functions to virtually emulate equipment but I don't have much experience with that system.

closed 2018-06-25 by pottedmeat7

Ubranek, 2018-06-25:
Yes, that right.
I had rtfm issue with InputEmulator and today found that out.
Thank you for your time.
Messed with projects a little.