pottedmeat7/OpenVR-WalkInPlace #35: Oculus Rift Support in the future?

galio094, 2018-06-25:
I can see that the tool is able to track Oculus Rift movement at the moment but not "moving" in VR. 

It seems it doesn't support Oculus yet? :)

pottedmeat7, 2018-06-25:
I am working with a loaned WMR system, I have found a few issues on why WMR doesn't work. Im thinking that maybe if I get it working for WMR that it will fix similar issues with the Rift. I have fixes for WMR headsets but I have noticed a few other bugs that I am working on fixing before I release again, maybe by the end of next weekend so I can test more...

galio094, 2018-06-25:
If there's anything I can help with lmk ;]. Really looking forward to it
and appreciate your hard work.

pottedmeat7, 2018-06-29:
Hey @galio094 I released the update for WMR, can you let me know if it also works for the thumbsticks on Rift?

galio094, 2018-06-30:
Looks like it doesn't work. Would me showing the graphs results it's showing on this end (video for example) help with anything?

pottedmeat7, 2018-07-02:
I think I have found the issue for thumbsticks vs the touchpad in the driver system. So that would probably stop the input from working for a thumbstick only setup... Was tracking working for the HMD values on the graph?

pottedmeat7, 2018-07-04:
Ok now that there is a distinction between touchpads and thumbsticks Oculus thumbsticks may work... Can you try the latest Pre-Release to check? If tracking is working then this input issue fix should make virtual input actually work...

galio094, 2018-07-07:
Hi There, Sorry for the delay coming back. It wasn't working for me but I
recorded a video (of the settings and what the graph is showing me) which I
hope will be of use.

https://www.youtube.com/watch?v=rDfmu87PwNY&feature=youtu.be

pottedmeat7, 2018-07-08:
Ok too bad. Looks like input is mapped differently for Rift. Also looks like there may be other tracking issues, it should be much more steady then that. There will be future progress done to support Rift still.

pottedmeat7, 2018-07-31:
Hey @galio094 Just watched this video again. Actually it looks like you are mostly using arm movement, your HMD also has to bob up and down. If you want to try just an Arm Swing you can lower HMD Y very low. I dont really want to make an arm swinging type of movement fully supported as I dont think its a realistic way to cause movement...

weissman20, 2018-08-06:
Hi,is working with wmr?i use but not see yellow lines.Great work with this addon

pottedmeat7, 2018-08-06:
Check that the graph looks close to this when walking in place https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/wip_graph.png 
WMR uses inside out tracking so its not as consistent and steady as the standard lighthouse tracking, at least with SteamVRs system. 
I had to reset the tracking for WMR, not sure if its because it needs to get a more accurate 'center' space or what... 
You could also try lowering the HMD Y values until it makes for steady step detection...

pottedmeat7, 2018-12-21:
Rift should work in the latest release. If anyone can confirm.

galio094, 2018-12-22:
Just tried the new version and it is not working from my experience.

I have un-installed the old version and installed the new one. Can see that
SteamVR detected it and is showing in the SteamVR Options. Pressed to auto
configure and I can see the graph moving and adjusting the values after the
setup. However when doing the movement in-game (onward) no "movement" is
being made in-game even the WIP showing clear activity in the graphs.

orcus0, 2019-02-07:
I just started with vr. I have downloaded this mod for Fallout 4 VR, but i cant find good instructions on how to set this up for occulus. When i watch the chart, i see the yellow step indicators. There is no movement in the game. i have been unable to find the right binding settings. I am also hitting a loading icon that freezes the game. it will load for 10-30 seconds every minute or two.  Perhaps one of your experienced users could write a walk through?

orcus0, 2019-02-08:
It is working. I will have to mess with the settings to get the speed and trigger points correct.  I am still getting the loading pause.  It is even happening in steam home, so it is probably an issue with steam?

closed 2025-06-08 by pottedmeat7