pottedmeat7/OpenVR-WalkInPlace #37: Speed getting stuck after changing settings

catbert7, 2018-07-07:
When I change some of the settings, the speed gets locked to jog or run for a while afterward and I can't switch to walking until it resolves itself. Haven't been able to determine exactly what resolves it but it doesn't seem to be time. If I just keep turning WiP on/off with the optional activation button (hold grip to activate, in my case) and moving at walk speed it starts working after 5-10 seconds (it will only do run speed until then). Again, a minor annoyance ;) If I figure out anything more about it, I'll update you.

pottedmeat7, 2018-07-21:
Yes noticed this bug also. It is now fixed in the latest release.

closed 2018-07-21 by pottedmeat7

pottedmeat7, 2018-07-22:
Ok just realized this still happens with running. Its difficult to test, especially because Skyrims sprint function, which causes other issues. But jogging slowing to walking does work now.