pottedmeat7/OpenVR-WalkInPlace #38: Suddenly stopped working today

Sledd00, 2018-07-26:
I'm using this plugin for Fallout 4 VR and The Forest, and I really love it.  Problem is, today it randomly stopped working.  Not sure what the issue is, but the graph shows the spikes and yellow lines, yet no movement translates to the game or wand.  I tried reinstalling it three times, tried the OpenVRInputEmulator version, I'm at a loss.  I can't really parse the logs either, but I'll attach them.  Any help would be greatly appreciated!
[driver_vrwalkinplace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/2233744/driver_vrwalkinplace.log)

sutekiB, 2018-07-27:
It stopped working for me too. This seems to have coincided with the latest SteamVR update. It was working just fine, and then when I played again a few minutes later, turning on on the Vive tracking pucks brought up a dashboard message saying "change bindings for..." - walk in place hasn't worked since, no matter what I try.

Sledd00, 2018-07-27:
That's actually fantastic news.  I thought I had messed something up on my machine.  Thank you for the feedback!

sutekiB, 2018-07-27:
I noticed you're not getting enough support, so I tripled my pledge on Patreon. It's still not much (I'm in debt like most people!), but I really appreciate what you're doing. This software makes a big difference to me, and to others I'm sure. I also shared your Patreon page on the forum I help moderate, so I hope that brings a few more in. Please don't abandon this!

pottedmeat7, 2018-07-27:
I'm so glad you enjoy the software!
I appreciate the support!
I have enjoyed building it and having it used
I will work my best to continue development and support the new versions of
SteamVR.
I think maybe after this next build, I will have a more stable input
solution.

pottedmeat7, 2018-07-27:
I will leave this open for the bug to update OpenVR input system.

pottedmeat7, 2018-07-31:
Ok I am on SteamVR production release branch and my system is now working. It also seems to currently work on the SteamVR Beta branch.

closed 2018-07-31 by pottedmeat7