pottedmeat7/OpenVR-WalkInPlace #41: VR server crashing when in Menus

Sledd00, 2018-09-16:
I'm sure this is user error, because it hasn't happened with any previous releases, but whenever I try to change a value in the menu, or look at the graph, I'm experiencing a screen freeze.  Steam says the VR server is no longer working and I have to restart my whole system.  I've tried uninstalling and reinstalling the drivers (this one and Open VR Input Emulator), reinstalling Steam VR, and playing around with every setting and bugfix I could locate, but I'm stumped.  Any help would be greatly appreciated!

pottedmeat7, 2018-09-16:
Ok it's good to hear it's not just my system. I did just notice this issue starting 9-14. They may have been another steam update. 
I think until I make the entire driver system using the supported methods then it's going to have these problems. 
I actually have been working on removing the unsupported features as im getting this to compile on Linux. 
So the next version is going to be a major update.

pottedmeat7, 2018-12-15:
Fixed in 4.0

closed 2018-12-15 by pottedmeat7