pottedmeat7/OpenVR-WalkInPlace #42: Track Pad input not working (OVRIE user)

TripRodriguez, 2018-09-26:
I saw another (closed) issue where this was mentioned.  I use OVRIE for my motion simulator motion compensation and don't want to uninstall that (which is how the other guy fixed this).  

Can you give some more info on getting WIP working when OVRIE is also installed?  As near as I can tell everything works fine except that WIP does not emulate the track pad inputs.  No grey dots on the virtual track pads and no movement in game.  Trying to use this for "Unknightly" ATM.  

Thanks

TripRodriguez, 2018-09-26:
Also note: I used the regular installer because I did not see an alternate installer when I followed the link to the downloads.

pottedmeat7, 2019-05-05:
This should be fixed in the latest release

closed 2019-05-05 by pottedmeat7