pottedmeat7/OpenVR-WalkInPlace #43: New steam tracker update, WIP not responding properly

upwn, 2018-10-12:
Hi, i've been using walk in place for a while now with and without trackers. It has worked fine with every game i've played up until just recently when steam reassigned the default input types for trackers. It almost seems like it's game related, as in the step graph i'm triggering the step smoothly. My profiles are all set up the way I want them and i've had plenty of experience using walkinplace. The forward movement happens when using the touch pad but when I try to trigger the walkinplace step it's stuttering forward slightly and not consistently.

I was recommended by a steam news article to disable the new tracker assignment in the steamVR dashboard settings if I want to continue using them like before. 

I've tried turning off and on beta for steamVR fallout 4. I haven't had the chance to try any other games yet but on my list that i've tried before is skyrimVR, Surv1v3, OrbusVR. I did try skyrim actually and there was lots of frame rate loss but I had just turned my settings up to high so that might be a reason.

upwn, 2018-10-13:
I have tried it again with skyrim and the same stuttering problem seems to be occuring. When I look at the controller after pressing the system button the trackpad display also appears to have a flashing dot. When I touch the trackpad the dot is solid. I have tried assigning the trackers to disabled, and left and right foot, but neither makes a difference. Hoping you can find a solution! walk in place is the only way I want to play.

upwn, 2018-10-13:
found another issue like this so i'll close this one

upwn, 2018-10-23:
I am still having an issue with this so i'm reopening. I've tried different set ups (moving shiny objects out of the room, smaller playspace, tracker dongles on usb extension cables closer to my playspace, and i'm sure it's not my set up or any issue with the trackers. The step graph looks as it should but on the controller in menu the dot is flickering opposed to solid when I touch the trackpad. The move input in game is similar to this flickering.

pottedmeat7, 2018-10-24:
Im interested to know whether you are on steamvr beta or not? There has been updates from steamvr both on non-beta and more in beta. I was also experiencing issues and am working on different methods of emulation to make the system more stable with new steamvr releases, but it may take some time.

upwn, 2018-10-25:
I've tried both steamVR beta and non beta and both are the same. I also tried assigning the trackers as feet, and as disabled like it recommends but both are the same.

The Keyboard/wasd option in the drop down menu works in Skyrim, unfortunately this is not really suitable, but it translates movement into key presses. Just something to do with the controller. When the dot is flashing on the controller at the walk input value it's also flashing at the center of the trackpad, I don't know if this helps.

generalsumthing, 2018-10-25:
im having the exact same issue.

upwn, 2018-10-31:
hey I didn't read your comment properly pottedmeat7. When I say that it moves me forward, i mean very slightly, if at all. It's unplayable. At the point that i'm at in fallout I have power armor. Because the dot is (or looks like) it's clicking in the center of the trackpad as well as at walk jog and run values on the trackpad, it's pushing me forward milimeters at a time and not consistently as if I were pressing the trackpad, and also exiting the power armor as if I were holding down center trackpad click.

generalsumthing, 2018-11-04:
when i am walking in place the button is being pressed at all times, but its constantly fluttering between the desired value and zero. 
same go's for jogging/sprinting.
this gives it a very jittery effect.
but in the step graph it runs perfectly.
ive tried uninstalling and reinstalling on 3 different hard drives.
same thing on all 3.

pottedmeat7, 2018-11-04:
Yes their will likely be issues getting this on running in every branch of steamvr right now. There is different issues with each branch but none will work currently. 
I have tried a few ways, and am able to get touchpad input via WIP however emulation of all the other buttons, grip, trigger etc. ceases to work, which likely is an un-acceptable release. 
There is also some noticeable changes to the Steam VR Input system that could completely change my approach on making this work. 
For the moment this project is essentially broken.

upwn, 2018-11-05:
Hey man that sucks, walk in place is my favorite thing about VR. I'd go as far as to say I can't go back to regular trackpad locomotion. If it helps next time I get some money i'll start making a small monthly contribution on your patron account.

Just a side note what do the other buttons need to do? I have only ever used touchpad input via WIP, and those options for other buttons to start or stop WIP are disabled. When I want to stop moving via WIP I just touch the touchpad and it stops accepting input from WIP. Usually the other buttons by default are mapped already to important functions in game and cannot be used with WIP anyway unless you remap the controller. Is it possible to do a slightly broken but mostly working release?

Unless you mean that walk and jog input work through touchpad input, but running does not work because you cannot emulate the touchpad click input. I wasn't sure if that's what you meant but if it's broken like that I guess it's broken.

pottedmeat7, 2019-05-05:
This should be fixed in the latest release

closed 2019-05-05 by pottedmeat7