pottedmeat7/OpenVR-WalkInPlace #47: How To Install?

Zverina, 2018-11-23:
No .exe I can clearly run anywhere in here and the Readme says to execute the latest installer.

What am I executing? Do I need a special program to run something in here?

closed 2018-11-24 by Zverina

Zverina, 2018-11-25:
Originally didn't see the very bottom about building and using NSIS.

Except I'm stuck again, attempting to build using Visual 2017 gets this:
`1>------ Build started: Project: lib_vrwalkinplace, Configuration: Debug x64 ------
2>------ Build started: Project: driver_vrwalkinplace, Configuration: Debug x64 ------
1>vrwalkinplace.cpp
2>DeviceManipulationHandle.cpp
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>c:\users\zverina\downloads\openvr-walkinplace-4.0\driver_vrwalkinplace\src\devicemanipulation\devicemanipulationhandle.h(4): fatal error C1083: Cannot open include file: 'openvr_driver.h': No such file or directory
2>dllmain.cpp
1>Done building project "lib_vrwalkinplace.vcxproj" -- FAILED.
3>------ Build started: Project: client_overlay, Configuration: Debug x64 ------
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>driver_ipc_shm.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>ServerDriver.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>WatchdogProvider.cpp
3>Moc'ing WalkInPlaceTabController.h...
3>The system cannot find the path specified.
3>Moc'ing overlaycontroller.h...
3>The system cannot find the path specified.
3>C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppCommon.targets(209,5): error MSB6006: "cmd.exe" exited with code 3.
3>Done building project "client_overlay.vcxproj" -- FAILED.
2>c:\users\zverina\downloads\openvr-walkinplace-4.0\driver_vrwalkinplace\src\driver\watchdogprovider.h(3): fatal error C1083: Cannot open include file: 'openvr_driver.h': No such file or directory
2>driver_vrwalkinplace.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>common.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>ITrackedDeviceServerDriver005Hooks.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>IVRDriverContextHooks.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>IVRDriverInput001Hooks.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>IVRProperties001Hooks.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>IVRServerDriverHost005Hooks.cpp
2>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\crtdefs.h(10): fatal error C1083: Cannot open include file: 'corecrt.h': No such file or directory
2>Generating Code...
2>Done building project "driver_vrwalkinplace.vcxproj" -- FAILED.
========== Build: 0 succeeded, 3 failed, 0 up-to-date, 0 skipped ==========`

Attempting to run the installer with NSIS gets me this:
`MakeNSIS v3.03 - Copyright 1999-2018 Contributors
See the file COPYING for license details.
Credits can be found in the Users Manual.

Processing config: C:\Program Files (x86)\NSIS\nsisconf.nsh
Processing default plugins: "C:\Program Files (x86)\NSIS\Plugins\x86-ansi\*.dll"
 + Banner::destroy
 + Banner::getWindow
 + Banner::show
 + BgImage::AddImage
 + BgImage::AddText
 + BgImage::Clear
 + BgImage::Destroy
 + BgImage::Redraw
 + BgImage::SetBg
 + BgImage::SetReturn
 + BgImage::Sound
 + Dialer::AttemptConnect
 + Dialer::AutodialHangup
 + Dialer::AutodialOnline
 + Dialer::AutodialUnattended
 + Dialer::GetConnectedState
 + InstallOptions::dialog
 + InstallOptions::initDialog
 + InstallOptions::show
 + LangDLL::LangDialog
[... 83 lines cut ...]
FindWindow: output=$0, class="Qt5QWindowIcon", text="SteamVR Status" hwndparent="" hwndafter=""
StrCmp "$0" "0" equal=+3, nonequal=
MessageBox: 48: "SteamVR is still running. Cannot install this software.
Please close SteamVR and try again."
Abort: ""
IfFileExists: "$INSTDIR\OpenVR-WalkInPlaceOverlay.exe" ? 0 : +5
MessageBox: 49: "OpenVR Walk In Place is already installed. 

Click `OK` to upgrade the existing installation or `Cancel` to cancel this upgrade." (on IDOK goto upgrade)
Abort: ""
StrCpy $upgradeInstallation "true" () ()
FunctionEnd
Section: "Install" ->(SecInstall)
StrCmp "$upgradeInstallation" "true" equal=0, nonequal=noupgrade
DetailPrint: "Uninstall previous version..."
ExecWait: ""$INSTDIR\Uninstall.exe" /S _?=$INSTDIR" (->)
Delete: "$INSTDIR\Uninstall.exe"
Goto: afterupgrade
SetOutPath: "$INSTDIR"
File: "LICENSE" [compress] 12112/35147 bytes
File: "..\client_overlay\bin\win64\*.exe" -> no files found.
Usage: File [/nonfatal] [/a] ([/r] [/x filespec [...]] filespec [...] |
   /oname=outfile one_file_only)
Error in script "C:\Users\Zverina\Downloads\OpenVR-WalkInPlace-4.0\installer\installer.nsi" on line 107 -- aborting creation process
`

skevv, 2018-12-05:
I too am getting this exact same error, and was wondering how to fix.