pottedmeat7/OpenVR-WalkInPlace #50: SteamVR For Linux

pottedmeat7, 2018-12-16:
Can rebuild the entire project on linux. 
The Overlay runs although dashboard input seems inverted it still works.

However after installing the driver in steamvr drivers path VRServer crashes, not sure of cause.

Maybe if someone else interested in playing VR on Linux and interested in this project can check out more. 
Using qmake and cmake you can rebuild the entire project using the provided `build.sh` and `run.sh` will run the Overlay.
Im copying all the driver resources and binaries directly into the SteamVR drivers path. I will make a script for that soon. 
Most of the steps are on the Readme. 

The issue here is determining if these issues with the Overlay Input and the crashing driver are due to OVRWIP system or just current SteamVR Linux support.

frostworx, 2019-12-10:
I recently received my Cybershoes, which only have windows-drivers. As their developer is following you, I found your project and just gave it a try
on arch linux I had to compile mlpack first.
after that I had to add 
QMAKE_CXXFLAGS += "-fopenmp" to your pro file as the compiler complained with
   51 | compiling without OpenMP support.  This will almost certainly cause \
   52 | irreparable disaster.  Either enable OpenMP support in your application (e.g., \
   53 | add -fopenmp to your compiler command line), or, recompile mlpack *without* \
   54 | OpenMP support."

I used system easyloggingpp-9.96.4-1
and pulled the openvr submodule from valve as submodule init didn't work
in WalkInPlaceTabController.cpp:1943 and following lines I had to replace
#if defined _WIN64 || defined _LP64
with
#if  defined (_WIN64)  || !defined (_LP64)
apart from this everything built fine.
I installed linuxdeployqt and created some "portable dir" with required libs and plugins
not sure if the qtdata dir (like in the windows installer) is required so I manually created one with linux versions inside. I put the whole dir into my SteamVR/drivers/ directory
not sure if I did something totally wrong but running the binary (both standalone and via 
~/.steam/steam/ubuntu12_32/steam-runtime/run.sh ./AppRun)
leads to this segfault:
OpenVR-WalkInPlaceOverlay: tpp.c:82: __pthread_tpp_change_priority: Zusicherung »new_prio == -1 || (new_prio >= fifo_min_prio && new_prio <= fifo_max_prio)« nicht erfüllt.

haven't debugged anything yet as of lack of time and as I'm not even sure if this is supposed to work like that :)
And sorry, that I couldn't even come close to help with your initial problem...
but hey,  a reply before issue's 1st birthday \o/

pottedmeat7, 2019-12-11:
This is helpful, thank you. I was just recently planning on starting to get this to work on Linux again. Last time I worked on it, I wasn't using mlpack, so I'm glad you figured how to build that!
I was also getting issues when installing and running the driver and overlay. At the time I was thinking there might just be issues with openvr drivers and steam VR for Linux support. 

I believe they have improved Linux support lately so I will also check it out soon and see if I can get any other results...

frostworx, 2019-12-11:
glad to hear, I could help a bit, and great that you're still interested in linux compatibility.
steamvr on linux is a moving target imho. all in all it is getting better and better, but it can happen once ina while that a (beta) update makes things worse.
Just let me know if I can help you any further.

jtgans, 2021-04-23:
So I've been giving the linuxoverlay branch a crack at running, and it seems like something really screwball is going on w.r.t. GLX, because when SteamVR is running, I can't start the overlay and get this error:

```
QXcbIntegration: Cannot create platform OpenGL context, neither GLX nor EGL are enabled
```

Oddly enough, if I use your run.sh with `LD_LIBRARY_PATH` set correctly, it starts up SteamVR, but then fails to start the overlay with exactly the same error.

UltraBlackLinux, 2025-08-09:
Trying to compile this today using `build.sh` I'm not having much success. The variable `isWindows` has never been defined and I'm wondering what that is even for. 

On top of that `driver_vrwalkinplace/src/driver_vrwalkinplace.cpp:35:33: error: ‘watchdogProvider’ was not declared in this scope`

Any idea how to progress here?
Thanks!

pottedmeat7, 2025-08-09:
I pushed some more changes for the latest Windows build. Which I think may have broken parts of the Linux build process again. There has been a few conflicts between the two platforms. If you revert back to the commits around September 9th ( 2a069853eb7fc241f14ff46d76651b0f82138fbf ). That build may work better on Linux, so something may have been broken when fixing the windows builds following that commit. Maybe it is clear what that breaking change was. Although I can't dedicate time ATM to find what it was. I think support for both platforms and having both working is getting closer, but still is a bit of a pain unfortunately.

UltraBlackLinux, 2025-08-09:
surprisingly the latest (after my modifications) and current branch fail to build with the following error:
```
lient_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp:561:29: error: ‘directory_iterator’ is not a member of ‘fs’; did you mean ‘directory_entry’?
  561 |                         fs::directory_iterator end_iter;
      |                             ^~~~~~~~~~~~~~~~~~
      |                             directory_entry
client_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp:562:34: error: ‘directory_iterator’ is not a member of ‘fs’; did you mean ‘directory_entry’?
  562 |                         for (fs::directory_iterator dir_itr(model_loc); dir_itr != end_iter; ++dir_itr) {
      |                                  ^~~~~~~~~~~~~~~~~~
      |                                  directory_entry
client_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp:562:73: error: ‘dir_itr’ was not declared in this scope; did you mean ‘d_ptr’?
  562 |                         for (fs::directory_iterator dir_itr(model_loc); dir_itr != end_iter; ++dir_itr) {
      |                                                                         ^~~~~~~
      |                                                                         d_ptr
client_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp:562:84: error: ‘end_iter’ was not declared in this scope
  562 |                         for (fs::directory_iterator dir_itr(model_loc); dir_itr != end_iter; ++dir_itr) {
```