pottedmeat7/OpenVR-WalkInPlace #52: Vive phantom controller, not working/idiot user :)

skevv, 2019-01-18:
Good Afternoon,

I have recently came across your WIP and seem to be having trouble with it.  Granted I know next to nothing about programming so I'm probably way out side my element trying to get it functioning correctly.  I'm not using any trackers, just the HMD, and controllers.

The issue I'm having is after installation my steam app is showing a 3rd controller.  I assume that supposed to happen.  The problem is while it can detect that there is a 3rd controller it will not pair, nor identify.  Consequently I'm having a phantom controller show up in my left hand.  No matter what game I'm playing the vive wand is displayed in my left hand, and flickers rapidly.  Fast enough that it looks like a solid grey controller.  When in the steam menus, the flickering stops.  The only way to fix this issue is to completely uninstall the WIP driver.

Also, I'm assuming that this because steam can't pair, or identify this 3rd controller this is what is preventing me from having any input in skyrim, and fallout.  I have been able to get it function in Arizona Sunshine, but its no where near usable.  In the few instances where I could get it working, I'd always run to the left like my left leg was 1/2 the length of my right leg.

I've jacked with all the settings, turned the controller off and on, reinstalled both the WIP, and Steam,  and spent hours trying to get this functioning properly.  Any help you can offer me would be greatly appreciated!

I've updated Steam, all my drivers, etc.  If there is any additional information you need let me know.

Thanks!

pottedmeat7, 2019-01-18:
Yes there should be a third controller shown in steamVR. You shouldn't need to attempt to pair or connect it at all. Even if its grayed out it should be fine. As you can see the flickering controller the controller is pairing correctly. 
The flickering is just an effect of SteamVR and these third controllers. Previously SteamVR displayed no more then two devices in any application, looks like they changed that. 
It sounds like everything is setup fine, more likely a tracking issue.
You are getting input into Arizona Sunshine, so that is a good sign that its functional.
What type of HMD do you use? 
If you are moving to the left, maybe the game is using controller direction and you are pointing to the left?
Is tracking consistent if you do more direct up and down squat style movements? Also swing your arms to jog/run while squatting.

skevv, 2019-01-18:
Hi Mr Meat,

I'm using the standard Vive HMD.  As for moving to the left, I made sure I didn't have my thumb on the touch pad, and was pointing both controllers in the direction I wanted to walk.  The actual moving mechanic was incredibly choppy and inconsistent.  Its hard to tell which part of the WIP isn't reacting the way I would expect it to because none of it is.  Sometimes I'll "run" but the game detects walking, sometimes I'll "walk" and the game will do nothing.  If you think of directional movement like a clock with 12 moving forward, and left being 9, when I move I seem to walk toward the 10 or 11 direction, it doesn't really matter which way point the controller, or look.

I've ran the auto-detect (I think that's what you call it) function to dial in the settings as well.  And I've changed them manually to really dial it in which has helped.  But it's far from usable for actual game play.  Right now its causing me to lose my balance.

Is there anything I can do to get the flickering controller issue resolved?  Also, for the bethesda games is there something I'm doing wrong?

One other thing I should mention now that I've thought about it, is the fact that bringing my controllers up to the HMD (like I'm adjusting the HMD), regardless of using the WIP, causes my HMD to crash, or zoom way out, and do all kinds of other crazy moving, flickering and blacking out.  So I wonder if there is some other tracking problem I'm not aware of.  If you have any insight into that it could be the root of my problem.

I'll run the WIP some more tonight, and try and provide you with more concise details to help with diagnostics.  I really appreciate your help.

Thanks you,

Kevin

pottedmeat7, 2019-04-08:
Looks like the flickering was a bug that never removed the overlay for the device. I fixed it somehow in the new release with double checking to clear the render model, maybe it was never cleared....

closed 2019-05-05 by pottedmeat7