pottedmeat7/OpenVR-WalkInPlace #55: Two controllers overlaid/WIP not working

cdibbs, 2019-02-04:
Gave WIP a try for the first time, tonight. It worked great for a few minutes, then after fiddling with settings and autoconfig, I noticed two things. One, it no longer registered any walking (Fallout 4 VR was the only place I tried and had it working). Two, an HTC Vive controller showed up superimposed over my Oculus touch controller (I've never owned a Vive). It was flickering as though the Vive controller rendered every other frame. My hunch is that it stopped working because the ghost controller was absorbing the output of WIP. 

Exiting SteamVR and then reentering clears the ghost controller (at least visually), but only until I enable WIP, again. Eventually, I narrowed the option that causes this down to the end of the second options row, where I select "2nd controller." Once I select that, the ghost appears. But, no matter what, walking has not been registering with FO4VR since those first few minutes when it worked.

Anyway, I've tried things like rebooting and removing/reinstalling WIP. So, I am not sure where to go from here. When WIP did work, it was very encouraging, so any help would be greatly appreciated. Thanks!

cdibbs, 2019-02-10:
FYI I revisited this, today, and after digging through the issues, again, I found this comment: 

https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/35#issuecomment-461628335

That got it partially working. However, if I try to improve upon the working setup via auto configuration, it stops working until I restart OpenVR, but at least there's no ghost controller. If I save a profile after running auto configuration, and then apply that profile, then I get a ghost Vive controller on my left hand.

Bugs, bugs. :-) I'd dive in to help, but my C++ is far too rusty (not the language, but the metal oxide). Thanks for your work on this!

pottedmeat7, 2019-04-08:
You shouldn't see the ghost controller in the new release

cdibbs, 2019-04-09:
Fixed, indeed. Thank you!

closed 2019-04-09 by cdibbs