pottedmeat7/OpenVR-WalkInPlace #56: The installation fails

dara-mtl, 2019-04-07:
The installation of the version 4.1 fails everytime.
-The installer can't complete the operation: "error opening file or writing: \drivers\00vrwalkinplace\driver.vrdrivermanifest" and it keep going on like this with other files.
-Ive tried to disable the antivirus/malware and restart the computer with the same result
-The v4.0 installer works fine

pottedmeat7, 2019-04-07:
Hi dara, the installer will give these errors if it can't run the exe and steamvr. 
Please go to `program files/Openvrwalkinplace /` and run the `startdesktopmode.bat` streamvr is likely not starting correctly.

dara-mtl, 2019-04-07:
Hi pottedmeat7, thanks for the quick reply! I've managed to make it run but I had to manually download and add the following dll to the program folder:
VCRUNTIME140D.dll
MSVCP140D.dll
ucrtbased.dll

With those in place, I could finally run the the bat and then the overlay and then a fresh install with the installer would work.

After restarting SteamVR and creating and applying a model, I could activate WIP but unfortunately it would not run. Here is the Overlay UI Log:
[INFO ] 2019-04-07 15:44:06: Application started
[INFO ] 2019-04-07 15:44:06: Log Config: C:\Program Files\OpenVR-WalkInPlace\logging.conf
[INFO ] 2019-04-07 15:44:06: Log File: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/VRWalkInPlace.log
[INFO ] 2019-04-07 15:44:06: Sound effects disabled.
[INFO ] 2019-04-07 15:44:06: Settings File: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace.ini
[INFO ] 2019-04-07 15:44:07: VR Runtime Path: C:/Program Files (x86)/Steam/steamapps/common/SteamVR
[INFO ] 2019-04-07 15:44:10: Found device: id 0, class 1, serial WindowsHolographic
[INFO ] 2019-04-07 15:44:10: Found device: id 1, class 2, serial MRSOURCE1
[INFO ] 2019-04-07 15:44:10: Found device: id 2, class 2, serial MRSOURCE0
[INFO ] 2019-04-07 15:44:10: unable to load data model at: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/default.csv
[INFO ] 2019-04-07 15:50:34: Application started
[INFO ] 2019-04-07 15:50:34: Log Config: C:\Program Files\OpenVR-WalkInPlace\logging.conf
[INFO ] 2019-04-07 15:50:34: Log File: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/VRWalkInPlace.log
[INFO ] 2019-04-07 15:50:34: Sound effects disabled.
[INFO ] 2019-04-07 15:50:34: Settings File: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace.ini
[INFO ] 2019-04-07 15:50:34: VR Runtime Path: C:/Program Files (x86)/Steam/steamapps/common/SteamVR
[INFO ] 2019-04-07 15:50:36: Found device: id 0, class 1, serial WindowsHolographic
[INFO ] 2019-04-07 15:50:36: unable to load data model at: C:/Users/Asean-It/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/default.csv
[INFO ] 2019-04-07 16:02:37: Received quit request.

pottedmeat7, 2019-04-07:
For WMR you should choose "other" on the HMD drop-down. 
Then check out the graph to make sure it's tracking at all.
You will have to create a data model before it will work also

dara-mtl, 2019-04-07:
Yes this is what I have done, I've chose "other" and I have created and applied a data model before activating WIP. Maybe the values for the data model are not right. It all look quite different from the previous versions, so I'll try to teak them tomorrow. The log keep mentioning WIP can't load the data model, but it mention default.csv, so I will also try to rename my data model this way and see if it changes something.  Could my profile from previous version be interfering as well? Thank you for your help, and your great work! I really love this plugin.

pottedmeat7, 2019-04-07:
If the data model isn't loading it probably didn't save right. Check the path in the logs of it's there. 
Might be a naming issue.It should import any csv files you have in that directory as options in the drop down.

tselling, 2019-04-07:
I am having same install problem.  Trying to run bat file gives VCRUNTIME140D.dll is not found.

dara-mtl, 2019-04-07:
Actually I can load the data model I have created and I see it in the directory so I start to think it might be a value issue. I'll do it all over again and make more test tomorrow.

dara-mtl, 2019-04-07:
> I am having same install problem. Trying to run bat file gives VCRUNTIME140D.dll is not found.

Try to manually add the missing dll in the folder and start the bat, then after you'll be able to complete the install.

tselling, 2019-04-07:
More than one dll missing.  Previous version installed and ran fine so not sure why would get this error now.

tselling, 2019-04-07:
I uninstalled current version and reinstalled previous version without any problems.  This version doesn't want to install.

pottedmeat7, 2019-04-07:
Okay you have given me something to work on. I will try to produce this issue in my side.
Can't think of immediate need for a new DLL...

tselling, 2019-04-07:
From what I am reading those are debug version of the DLL's  so maybe needs a recompile in release mode.

pottedmeat7, 2019-04-07:
Also dara, if the file is there. Open it up and see what it's got in it. Should have about 10 lines total.

pottedmeat7, 2019-04-07:
Getting the tracked devices to load as checked as default was a bit of a problem that I had, maybe that's why... Once you save a profile it works too...

pottedmeat7, 2019-04-07:
The DLLS your missing are part of the visual studio redistributable.
It installs as a step off the installer, something must be going wrong in the install logs maybe..
You can download it also from here https://www.microsoft.com/en-us/download/details.aspx?id=48145

pottedmeat7, 2019-04-07:
I suppose I may have built mlpack in debug mode, and that's why... However I do not have these DLLS directly in the install directory.

tselling, 2019-04-07:
Yes, but you if you have visual studio installed for developing they are part of it so you will have them.  Most users will not have visual studio installed and so will not have the debug versions of those DLLs.  So you should include them or build in release mode.  Those DLLs are not in the user versions of the Visual C++ Redistributable packages.

pottedmeat7, 2019-04-07:
Yes looks like I build mlpack in debug, so its those looking for debug versions, I am recompiling those and should need no other changes. Coming later today

pottedmeat7, 2019-04-07:
Ok try should use the release vs DLLs now here it is https://github.com/pottedmeat7/openvr-walkinplace/releases/tag/4.1 
You can close this issue if it works, thanks

tselling, 2019-04-07:
New version installs without any issues.

tselling, 2019-04-08:
Working great in War Dust with trackers on ankles.  First time I have managed to get something with trackers working acceptably.  Thanks!

closed 2019-04-19 by pottedmeat7