pottedmeat7/OpenVR-WalkInPlace #57: I keep moving even when I stop walking

tselling, 2019-04-08:
I have found that I keep moving for a bit after I stop walking. Not for very long, maybe a second or 2. This is using trackers.  Not sure is there is a setting for this?

pottedmeat7, 2019-04-08:
I will put this as a feature request.
You may try changing the minimum touch pace to something lower, it will make it feel much better as a slower value, as low as 0.15 in Skyrim for instance.
There are limitations however
Due to the sample size required to check the current state of the device. I only run the process every 1 / 40th of a second. 1 whole second is too slow to use as a realtime decision, so I have to make a decision by 1/4 of a second, so I only get 10 numbers for 1/4 of a second. 
To know if the device has stopped moving I check if the values add up to less then the allowed variance. 
The reason that seems to keep it moving for a little while, is that one of the numbers in the set still adds up to more then the variance.
There could be changes made to increase the sample time, which would slow processing and use memory, or decrease the necessary sample set to less then 10 numbers which may decrease accuracy. Or decrease, or increase, the variance (this would probably work best...). 
I may be able to make it customizeable, but it would be difficult to customize the time step to something other then 1 / 40th of a second.
All these options are a bit difficult for me to picture on how to support, and if it would work better or not...
I think for some of these options it may be easier if the user edits and recompiles the project and see if they like it better. 
Its too much iterative testing for me ATM, but hopefully more progress to be made.
 
Just an explanation as to why this probably won't happen very quickly.

tselling, 2019-04-08:
Ok, thanks.  I may take a look at this as I am a semi-retired software engineer.

tselling, 2019-04-09:
When you get a chance, can you point me to the files where you have the variance, sample size, etc.  Thanks.

tselling, 2019-04-09:
Great thanks!  I hope to get the walking thing better for War Dust.  Will probably not be able to look at compiling until tomorrow or next day since expecting a new (to me) laptop today. The extra walking is really tough in a large multplayer shooter like War Dust.

tselling, 2019-04-16:
Hi, I finally got the project to build.  But missing LICENSE, *.vrmanifest, qtdata from the directory NSIS compiler looks for in $(OVERLAY_BASEDIR).  Not sure if something didn't build correctly or if you had copied some stuff in manually.  Please let me know.  Thanks.

pottedmeat7, 2019-04-19:
Don't know why but those files never were versioned in git. I added them. 
I was thinking about this and I think the best way may be too separate the decision for stopping from the main decision on the model.
The check for stopping just matches essentially to zero. The time step does not really matter, so that check could run at any time step separately from the model check.

tselling, 2019-04-19:
Thanks.  I finally got everything to build, install, and run.  Trying to
understand the code now.  Not sure why but the controllers have never
triggered walking for me... HMD and trackers do.  I am mainly focused on
trackers but just using the controllers sometimes would be nice, but I
don't like arm swing and would prefer to use up/down motion of controllers
similar to trackers.  Do you count all movement for controllers (including
vertical)?  Or just one direction.  I am new to quaternions so some of it
is rather confusing.

pottedmeat7, 2019-04-19:
Yes the controllers will not trigger walking by themselves currently, but requiring no HMD movement is more like arm swinging which I think is more unnatural.
You may also be able to do this if you enable your two controllers as "controllers on feet" in the enable device page. That allows you to basically use your controllers as trackers, normally for feet but you can record a data model with them in your hands; It doesn't really know the difference they will still behave like a pair of trackers.
I only utilize the Up and Down (Y axis) velocity of all the devices, this is the most significant data point to record and measure.
Yes that is where most of the main loop is.

tselling, 2019-04-20:
Ran a few tests.  I tried both halving the variance for the trackers to .35 and doubling to 1.4 and neither helped with the keeps walking problem.  Halving seemed like I kept walking a bit longer.  Doubling was harder to start walking and still kept walking after.   
When I enable the hmd and the trackers, seems like that works fairly well to make me stop (so code for stopping with hmd is works) .. but I have to make a conscious effort to bob my head while walking with the trackers.  Enabling the HMD rotation (not exactly sure what that setting is for) makes it so I never walk with the trackers so have to leave that disabled.
Using the held controllers for walking probably would never work too well for me because every time I duck I seem to start moving which makes sense since I move my arms down to when I duck.
I like your idea of moving the stopping logic for the trackers out of the main decision area but I don't know enough about how the main decision loop works to do that just yet.  I will continue to analyze it to see if I can figure it out.

pottedmeat7, 2019-05-06:
I pushed some changes I was trying out. I made a function to filter out the peaks and find the minimum and maximum peaks of a sample. I used this instead of the sum of the sample to track when stopping occurs.
Basically if the maximum peak of the last section of the sample is smaller then the minimum peak in the model, then it stops. 
There is still a delay when stopping though. The peak just doesn't get detected, or maybe even occur, until about a 1/6th of a second or so, which is noticeable, but its going to be difficult to get the data and make a decision much faster then that...

EpochalEngineer, 2023-11-19:
As of the 4.33, I have a rather significant 1-2 seconds of lag/inertia on my Quest 3, but will try playing with variance/ongoing/max size options.

pottedmeat7, 2023-11-20:
Great your trying out modifying the variances. Its possible there is a better set of those values, but not entirely sure how much better it can get. 
Using the controller model recording may be an effective way to improve the inertia problem, you may prefer to disable the "walking" detection, and input 0, ie. to not move. So it may detect "walking" before it completely stops input, but either way your character will stop moving.
But likely it's not 100% synchronized timing to when you actually stopped, so there's always some lag. 

You can also use the "Disable WIP" features where you can choose a button from the drop down when that button is clicked it will stop you from moving. Or also only enable WIP when a button is held.