pottedmeat7/OpenVR-WalkInPlace #58: WMR headset not working with latest release

godzan, 2019-04-11:
Hi, I tried this with my WMR headset and it looks like it is tracking, but there is no movement. I tried this with my Vive, and Rift, and I am able to get movement with those, although I can't get the run function to work on either of these headsets.

pottedmeat7, 2019-04-11:
Are the controllers showing tracking data on the graph? 
Also maybe check out how the steamvr's input bindings are setup for the ovrwip controller. You can change that as needed, it may change of you are using WMR or RRift controllers just rebind the ovrwip touch pad input to joystick etc.

godzan, 2019-04-11:
The controllers are tracking on the graph.

I'll have to look into the input tonight. I'm not quite sure how to do
that.

dara-mtl, 2019-04-21:
Got the same problem as godzan with a WMR headset. I can create a data model without problem and apply it, but if I go on the tracking data page I only see occasional red line for the headset no real-time tracking with green line. I try to create a new model several time, but to no avail.

For testing I always use Gorn and Dreadhalls has the same controller scheme as Oculus and Vive. All the previous WIP versions were working, but this one doesn't pick any values. Please tell me if there are any datas you need from me to help you.

pottedmeat7, 2019-04-21:
Uncheck "track HMD rotation" on the devices page. Maybe the rotational values are different in WMR...

godzan, 2019-04-21:
I unchecked "track HMD rotation" and it is still not working on the WMR Headsets I tried.

pottedmeat7, 2019-04-21:
Can you send me your csv data model file in the AppData directory.

godzan, 2019-04-21:
I'm not sure where that is located.

pottedmeat7, 2019-04-21:
It's here C:\Users\<USERNAME>\AppData\Roaming\pottedmeat7\OpenVRWalkInPlace\
There should be some csv files in there

godzan, 2019-04-21:
I cant upload it because Github doesnt support that file type.

godzan, 2019-04-21:
[le1.zip](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/3101425/le1.zip)
here it is the csv data model

pottedmeat7, 2019-04-21:
Looks like there is no HMD data in the model.
I only see controller data. Do you see data when you "show HMD" on the model page? 
Maybe something is in the log, it's in the same directory as the models.

godzan, 2019-04-21:
[VRWalkInPlace.log](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/3101508/VRWalkInPlace.log)

I am seeing data on the model page, but it is not as pronounced as my Rift or Vive.

pottedmeat7, 2019-04-21:
That's probably why. If the HMD values are not over a certain amount, it might be closer to what matches the stopping model, which is basically just a zero line. 
I'm not sure what value it would have to be over to match one over the other but under a certain amount and it will presume that it's just not moving.
I would recommend disabling the HMD tracking in this case, and user controllers as trackers, but it seems that feature isn't working completely. 
With this information I think I can add some support to fix these issues in the next release.

dara-mtl, 2019-04-22:
Same as mentioned by godzan, I also tried disabling the HMD rotation. I attached the csv as well.
[dara.zip](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/3101834/dara.zip)

pottedmeat7, 2019-06-16:
The latest release should help fix this issue. The data won't have to be as pronounced to cause movement.

godzan, 2019-06-16:
I can now move in the game, but I basically have to jump up and down in real life to activate movement.

dara-mtl, 2019-06-16:
Hi pottedmeat7, I've tried the latest version and made different tests, but I still can't make it work. Once I set up everything there are no reactions from the controllers. I can see the datas created by my controllers in the data model but I can't see yellow lines on the graph. I've tried with the HDM enabled and disabled. I think it might come from the initial setup; in the previous version, the automatic setup would but my controller value to 0 regardless of what I was doing, but then I could manually adjust it. I also looked for the OpenVRWalkInPlace.ini, but it is missing. If I create one and place it in the path you provided, it bugs the SteamVR Overlay.

Lenovo Explorer
Windows 10 1903  18362.175
SteamVR non beta version 1.5.9
WMR Portal on latest version

dara-mtl, 2019-08-19:
This release still doesn't work with WMR. I can create a data model without problem and I will see the lines of both my controllers and HMD, but if I check the "show data model" section, I will see a weak input for the HMD (in red) and no input on either controllers. I always use Dreadhalls as a test and set the options to "others" and "touchpad" before enabling WIP. It use to work great up to version 4.0, but now I can't move whatever I try. Godzan mentioned it was possible to move if we jump up and down, but I start to think WIP is just catching the HMD movements, not the controllers, thus moving a little if we start jumping.

I've tried with HMD enable and disable as well as with controllers enable and disable to no avail. I also try making several data model walking different way. Please let me know if you need additional information.

pottedmeat7, 2019-08-19:
Can you send me the overlay log and the driver log. Their locations are in
the readme

dara-mtl, 2019-08-20:
[Logs.zip](https://github.com/pottedmeat7/OpenVR-WalkInPlace/files/3519513/Logs.zip)
Here's the log files, I also added 3 different test profile I did, if it can help. Thanks!

pottedmeat7, 2019-08-28:
If you aren't getting tracking data for the controllers on the graph. Check to see your controllers are selected in the "tracked devices" page. Im suprised WMR controllers wouldn't output the same tracking data Vive controllers do, but maybe not...

dara-mtl, 2019-08-29:
Yes they are selected, controller 1 and 2. They generate lines and I can create a data model with them. But once the data model is created it is then impossible to have their movements match the data model, as if WIP could only write a data model using the WMR controllers but not read it. The headset will produce a red line, but the controller won't do anything, I've tried to "walk" as I was doing with the previous versions, I also tried more exaggerated movements, but they can't see to be able to match the movements from the data model.