pottedmeat7/OpenVR-WalkInPlace #59: Can't move in vr

Werschon, 2019-05-24:
Hello pottedmeat7, I can't get WalkInPlace to work. I want to use my two trackers to walk in The Talos Principle VR with my own feet. So I created a datamodel and when I click on 'show data model' the graph shows me that my trackers and my headset are moving but when I want to move in vr nothing happens.
[image]
[image]

These are my settings:
[image]

Then I tried to fiddle around with the ovrwip controller. As I understand it, ovrwip is a digital controller that receives the input from the trackers and forwards them to one of my Vive controllers, is that correct? And I have to use the steam mapping system to make sure that the input is indeed forwarded to one of the vive controllers, right?
So I checked my ovrwip controller and it seemed like the input should go to my left vive controller:

[image]

So by my understanding everything should work, but unfortunately it does not. Did I miss some important step? Or is it because I use a Pimax 5k+? I also noticed that the graph does not show the movements of my controllers:

[image]

Could this be the issue? Would be nice if you could help me, thanks.

pottedmeat7, 2019-05-24:
Looks like the tracking data did not get any output for the HMD. There should be some data similar to the tracker model there. Must be an issue with Pimax 5k+ sending incomplete openvr tracking data, may run through thier own software instead. 
By default the HMD model is required, however you can disable tracking HMD entirely on the disable device page. This will match just the trackers but is not working ideally ATM, it wont be as instant or smooth movement. The next release you will be able to customize the size and duration of the tracker real-time samples which will help make tracker only movement better. I would recommend HMD tracking also but looks like its not possible with pimax 5k+ ATM.

Werschon, 2019-05-24:
Thanks for your fast reply. Without HMD tracking it finally works, although not really good. But this is strange, the tracking graph page shows that the HMD is moving:

[image]

But the datamodel shows just this flat line

pottedmeat7, 2019-05-24:
Ok looks like the pimax maps over rotations but the velocity is not, actually this is similar to issues with the Rift tracking. Did you try to select "Rift/Other" from the dropdown next to "Enable WIP"? Maybe then you will get the Y_VEL values needed for the HMD model.

Werschon, 2019-05-24:
I tested both options ('Vive/Rift' and 'Other') but unfortunately Y_VEL still remains a flat line.

pottedmeat7, 2019-05-24:
Ok, this was a problem with Rift because it wasn't getting the direct velocity values available through openvr. Fixed by using the change of positional data, but that may not be available either with the pimax. The transformation maybe only contains the rotational data for pimax.