pottedmeat7/OpenVR-WalkInPlace #60: openvr issue that affects input

pottedmeat7, 2019-08-04:
https://github.com/ValveSoftware/openvr/issues/1153
The newest release of steamvr input has broken the virtual input but they say a fix is coming in 1.7 
There is a very handy new feature to test the input though.
You can go to the SteamVR dashboard -> Settings -> Controller Bindings -> Test Controller
You can see all of the active controllers and see if input is being applied.

pottedmeat7, 2019-08-18:
This is fixed and working in the new release 4.3 and the SteamVR beta

closed 2019-08-18 by pottedmeat7

peroht, 2019-08-20:
Tried this with our device with openvr 1.6.10b but our device does not show up. In fact the whole settings window stops working and I have to restart SteamVR to make it work again. I can configure the bindings for the controller however.