pottedmeat7/OpenVR-WalkInPlace #61: does it work with opencomposite?

caoHenriques, 2019-09-01:
congratulations for this amazing work,

does it work with opencomposite?
it would be great getting it working with opencomposite (https://gitlab.com/znixian/OpenOVR/blob/master/README.md)
because the  steamVR hit on performance is huge

thanks

pottedmeat7, 2019-09-01:
You mean the performance is bad on SteamVR when using Oculus, a non open hardware.
The initial decision that creates this problem is Oculus, or Facebook, I do not support that company.
This project will be built for openvr, and as well will be for OpenXR when it comes around and is functional. 
This project will work with closed hardware, but only because SteamVR and OpenVR work with that hardware.

closed 2019-09-01 by pottedmeat7

caoHenriques, 2019-09-01:
thank you very much for the reply.
I agree with you. Unfortunately who have a rift S cannot avoid using oculus software because Steam does not support rift S natively. 
As far as I know opencomposite makes the interface between OpenVR and the oculus drivers, avoiding the Steam layer, but with less performance hit, so I thought that since your software works with OpenVR it could also work with opencomposite. 
I hope OpenXR will solve these problems. 

cheers