pottedmeat7/OpenVR-WalkInPlace #64: Keyboard input for naming models/profiles does not work

doxxx, 2019-12-08:
When I go to create a new model or profile and it asks for a name, I can get the SteamVR keyboard to popup and enter a name but, after pressing Done, the name does not get entered into the model/profile name field.

Since this is the first time I'm using the app, this is preventing me from creating a model or profile and thus I cannot use the app at all.

pottedmeat7, 2019-12-08:
If the SteamVR keyboard is not working for input, which in SteamVR beta 1.9.7 I can confirm it isn't, you can launch the interface in desktop mode with the batch script `C:\Program Files\OpenVR-WalkInPlace\startdesktopmode.bat` 
Then you can enter text fields with normal keyboard. 
I will look into the VR keyboard bug and see if theres anything I can do. Or SteamVR will hopefully have a fix for this problem before releasing the beta...

doxxx, 2019-12-08:
Thanks, starting it in desktop mode let me create the model and profile.

pottedmeat7, 2019-12-21:
This seems to be fixed in the latest SteamVR beta

doxxx, 2019-12-23:
Yes, confirmed. Thanks.

closed 2019-12-23 by doxxx