pottedmeat7/OpenVR-WalkInPlace #65: More documentation of what the configuration options mean

doxxx, 2019-12-08:
I find the configuration screen very hard to understand, particular the check boxes and drop down lists in the top section. Could you provide more documentation on what each of these controls do and how they interact with each other?

It would be also very useful to provide more examples of configurations for particular games and controllers, e.g. SkyrimVR with Vive Pro HMD + Index controllers.

pottedmeat7, 2019-12-09:
The very top line of configuration for the most part is not needed anymore. I believe support for Rift and WMR has become similar tracking, at least as much as possible, so there might not be any difference for HMD selection. 

For the input options, doesn't really matter as whichever you choose, you have to create SteamVR bindings for the input anyway, it can be mapped to whatever you like. It's really just the "click to sprint" option that matters. 
All of that input is applied to the virtual "ovrwip_controller" device called "ovrwip_000" 

I hope that the second line of configuration is also irrelevant, because optimally you shouldn't need to hold any button to enable/disable the walk in place detection. That is the goal anyway, its left there in case you are playing some game where you make alot of HMD or arm swinging movement and are getting false positive movement. I don't have that problem in SkyrimVR or No Mans Sky, or other games I have been playing, but I leave it there in case others have that problem. Again optimally you would leave the setting as "disable/enable" with "only with data model" the controller used is irrelevant. 

What I think needs to be documented is properly setting up SteamVR input bindings. I don't know if there are many other good walkthroughs on setting up SteamVR bindings, but for the "ovrwip_controller" its the same process, however whats really important is that in "extra settings" you bind the input to the left or right controller.

doxxx, 2019-12-09:
I appreciate the reply, but you didn't really answer my question. Let me be more explicit:

* On the first line:
  * Do I need to check "Enable WIP?" Your example configuration image shows it unchecked.
  * If I have a Vive Pro HMD, do I set the first dropdown to Vive? What does this option actually do?
  * What does setting the second dropdown do? What does selecting "touchpad (click sprint)") do?

In the SteamVR bindings for the OVRWIP Controller:

* What does "return bindings" mean?
* Do I check the option for left or right or HMD? Or some combination of those three?
* Do I need to remap all the normal index controller buttons in the "Legacy Bindings (Handed)" tab?
* Or does my normal Index controller bindings work as usual along with the OVRWIP controller bindings?

pottedmeat7, 2019-12-09:
The details of these options are in the Readme starting here https://github.com/pottedmeat7/OpenVR-WalkInPlace#enable-wip 
Check Enable WIP, select Vive, use touchpad, and click to sprint if your game doesn't have another use for the click function.
I made some screenshots of setting up the SteamVR input bindings, available here:
https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/steamvr_bindings/Readme.md

You check left or right, whichever hand you use for locomotion input normally. You do not need the HMD option.
You do not need to remap all buttons just the touchpad.
The Index controller bindings are separate, you don't need to change them and they will work alongside.

doxxx, 2019-12-09:
So does that mean OVRWIP always sends touchpad input and I have to use the SteamVR bindings to map that touchpad input to the correct input for the game? e.g. map touchpad to joystick for index controllers in skyrimvr?

pottedmeat7, 2019-12-09:
Yes touchpad is all the simulated input needs. Then that can be mapped to anything you'd like.