pottedmeat7/OpenVR-WalkInPlace #68: Walk In Place Custom Controller (ovrwip_controller) does not appear as an option in SteamVR controller bindings menu.

chars709, 2019-12-21:
Hi pottedmeat7,

New user here. I'm trying to get your wonderful mod working and I can't tell if something is missing or broken, or if I'm just a dummy.

I'm following along with your picture guide, and when I get to the step where I pick the WIP controller, it's just not there as an option.

Thanks for all your great work :)

pottedmeat7, 2019-12-21:
Hey chars, very glad your enjoying the project!

pottedmeat7, 2019-12-21:
Hey chars,
Thanks for trying out the project!
If the ovrwip_controller is not showing in steamvr its likely it hasn't been created.
The ovrwip_controller will only be added after you have the number of controllers that you selected (either 1 or 2) have been turned on. After that you need to have loaded a complete data model record of your selected tracked devices. 
So create a data model first, with HMD, controllers, etc. You can save that to a profile so you don't have to re-select it each startup. Then you check "Enable WIP" once everything is setup, it will create the virtual "ovrwip_controller" you will see a extra controller added like this 
[image]
Really the virtual controller is the last step once all of your tracked devices, model, and settings are setup. Then you can create the steamvr bindings.