pottedmeat7/OpenVR-WalkInPlace #70: #USERTREADMILL shows no input on test and SkyrimVR won't take input - Oculus Quest on Virtual Desktop

Ravenwynne, 2020-04-01:
I performed all the configuration steps, ovr_wip shows as 000.  I set the bindings to your custom profile, left hand return.  When I go to test controller no matter what I do the test won't light up the USERTREADMILL controller buttons at all, and Skyrim VR is not seeing any movement.  Would love to chat you up on discord to see if I can get it working (will make a small donation if you can help me get this working on SkyrimVR so I can finally enjoy the game).

pottedmeat7, 2020-04-02:
Are you on SteamVR beta? What HMD are you using?

Ravenwynne, 2020-04-02:
Oculus Quest through Virtual Desktop.  I had it on SteamVR Beta, but I've since opted out of beta and it still hasn't worked.

pottedmeat7, 2020-04-02:
I don't know how the Quest tracking data is transferred into SteamVR, I think it takes another driver, which may interfere with this driver... 
They only way I know a quest would work, is with SteamVR lighthouses and a HTC tracker. Then you disable HMD tracking.

Ravenwynne, 2020-04-02:
The tracking data seems to be fine because I can create a model and see
tracking data on the graph.  I believe Virtual Desktop emulates an Oculus
Rift S (or whatever driver the Oculus software now uses for Steam VR).

For all intents and purposes, you can pretend I own a Rift and it's just
not working even though I can see and record tracking data.  Any ideas?

Ravenwynne, 2020-04-02:
I tested again and can confirm that, while after I check the ENABLE box I can see and select the active controller device OVR_WIP in the legacy settings, my Steam VR overlay on my PC desktop is only showing my HMD and 2 touch controllers.  There's no 3rd controller lit.

pottedmeat7, 2020-04-03:
When you look at the HMD data model with enabled checked. Do you see any green or red lines?

Ravenwynne, 2020-04-03:
You're right.  I don't see Y Vel green line for the HMD, so it might be
tracking differently than you had set the system up for.  I tried setting
the headset to "other" and still didn't work.

pottedmeat7, 2020-04-04:
Sounds like the same problem here https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/62
I had this issue with Rift not supplying the vVelocity data and it was always 0. But later it seemed to start working after an update by SteamVR or Oculus SDK...
The Other setting just tries to determine the velocity by change in position. If it doesn't work then that data is also not available.
Im not sure if there is anything I could do, unless I really research how those drivers work, I don't have a Quest.

Ravenwynne, 2020-04-05:
I really wish you owned a Quest.  No idea why it later started working.  Do
you still have a Rift and can test?

Ravenwynne, 2020-04-11:
I tested the new version you released that calls one HMD Steam VR and the
other Other.  Blue and Red Pitch and Yaw work, but Y_Vel is still being
stubborn and not working.  I will ask ggoddin the dev of VD and see if he
has any insight.

pottedmeat7, 2025-09-27:
This may be solved with the latest release.