pottedmeat7/OpenVR-WalkInPlace #71: Tracked Devices not saving properly/glitching out

Kingstyler88, 2020-05-07:
So I am currently trying to get this to work with two trackers on my feet.

But the problem is, that I have almost no control over which devices are being tracked. 

I go into the Tracked Devices page, tick "tracker 1 on feet" and tick "tracker 2 on feet". I then hit the back arrow key on top and as soon as I go over to the Tracking graph my settings didn't transfer over. So sometimes i will only have one tracker on my feet active, despite having enabled both. If I then check the tracked devices page again, it shows a completely different arrangement of trackers than that which i setup 5 seconds ago.

This issue occurs even if i go right back into the tracked devices page.

So I set it up as follows

Controllers in Hands:
Deselect Controller 1
Deselect Controller 2

(since i want it to only track my feet)

Trackers on Feet:
Select Tracker 1
Select Tracker 2

Hit the back arrow button in the top left corner

Go back into Tracked Devices and I get

Controller in Hands:
Controller 1 Selected
Controller 2 Deselected

Trackers on Feet:
Tracker 1 Selected 
Tracker 2 Deselected

The arrangement seems to be literally random, what could be causing this?

pottedmeat7, 2020-05-08:
I experienced this problem, and it was because of a conflict between the setting changed and the current profile. 
I don't remember if it was just the 'default' profile.
To fix it you have to not have a profile selected, then change your setting and save that as a profile.
You can also just edit the profile directly in this file
C:\Users\<USERNAME>\AppData\Roaming\pottedmeat7\OpenVRWalkInPlace.ini

pottedmeat7, 2020-05-08:
Also just a heads up, the data sample that occurs for trackers on the feet is actually quite complicated to track (there is a peak, followed by a long pause while the other foot is up).
You may also have to change the duration of the sample of the trackers in this file
C:\Users<USERNAME>\AppData\Roaming\pottedmeat7\OpenVRWalkInPlace.ini
I have had such good results with just HMD and controller tracking, that I don't use trackers anymore, so I haven't really been able to focus or test the tracker quality much.
It might actually work better on the waist or a belt or something...

pottedmeat7, 2020-07-12:
The issue with modifying a profile and it reloading the old settings should be fixed now.

closed 2020-07-12 by pottedmeat7