pottedmeat7/OpenVR-WalkInPlace #74: Character is still moving even after stop swinging controllers

CFA44, 2020-10-05:
Hi, 
I'm using OVR-WIP with Odyssey+ with swinging two controllers. 
My problem is the character is still moving even after stopping swinging controllers and it looks like that the character slides.  Please, see the following captured image:

[image]

I've installed OVR-WIP v4.32, released 27 Jul. 
Also, my OVR-WIP is set as:
* touchpad
* by holding trigger
Other setting are looked as the followings:
[image]
[image]
[image]
[image]

How can I fix that? Thanks in advance and the cool project!

pottedmeat7, 2020-10-05:
Looks like when you are recording the data model, you stop swinging your arms between transitions. 
Ex:
https://raw.githubusercontent.com/pottedmeat7/OpenVR-WalkInPlace/master/docs/screenshots/hmdSampleEX.png

Even between steps during recording you should just continue to walk in place (or swing arms etc.) to constantly record valid movements. There should be no still breaks in the model. 

Also you may need to change the settings in the .ini file for the trackers (or in your case controllers)
(the ini file is at `C:\Users\<USERNAME>\AppData\Roaming\pottedmeat7\OpenVRWalkInPlace.ini` ) 
change 
`walkInPlaceProfiles\<profile_index>\ongoingTRKRSampleSize=8`
to something lower, with a smaller number of samples it will make the required sample from the controllers shorter in time.

CFA44, 2020-10-06:
Thanks to comment.
I have solve my problem.
I had recorded another data model and change .ini value, ongoingTRKRSampleSize=3 because I need to react quickly swinging arms. It must be very low but check this captured image.

[image]

It was better motion than previous setting
I think most important setting is don't stop between steps when make data model. 
Thanks to advice.