pottedmeat7/OpenVR-WalkInPlace #75: Walking direction problem

CFA44, 2020-10-22:
Hi,
I'm using OVR-WIP with Vive pro with controllers and trackers.
It still good working.
But I want to change the direction of walking, not the controller direction.
It was the same even if I chose a different binding option. ( Return bindings with hand and hmd option )
Can I change it based on tracker? especially waist tracker.

Second,
Can I create and run data values ​​without a controller?
Turning off both controller tracking options and checking the Tracker option on the foot will terminate the plugin or reset the option. I need to walk or running without controller because I have to using simulated firearms and trackers.

Turn off two controller tracking and automatically turn on the one controller option when creating data. The same goes for setting the two trackers on the ankle with hands or feet.

pottedmeat7, 2020-10-23:
The input that's applied too the bindings only will apply 0-1 on the virtual touchpad without any left it right direction. Currently there is nothing else that determines a change in that direction. 

Usually the games have different "direction" settings, mostly either Controller direction or HMD direction.

You should be able to use just trackers. If the options you select are resetting, maybe the current settings in the profile that's asked are overriding what you've selected.

You can either select a different profile, delete and create a new one. Or you can manually disable the attached devices directly in the `.ini` file. 

You have to have 1 device defined as a "controller" however it can actually be a tracker too. I'm pretty sure this works, I usually use 1 controller, so I haven't tested that combination in awhile..

CFA44, 2020-10-23:
Thanks for the reply

However, if I set it as shown in the picture and run the tracking graph, the driver is forcibly terminated.

[image]

Is this a problem when using without a controller? Is a controller required for locomotion tracking?
Would it be better to modify the ini file and use it in case a problem occurs with these options?

Two controllers, two trackers, and HMD tracking options are working fine.

pottedmeat7, 2020-10-23:
I'm not sure why it crashes...
Try also selecting the same trackers ie. 1 and 2 (or 1 or 2) for"trackers in hands". That means the trackers will be the "controllers".

CFA44, 2020-10-23:
Your explanation was correct

[image]

When both hands and feet were checked, data was received normally.
I'm still not sure why the WIP plugin was forcibly turned off. In common, it seems to happen when all options related to the hand are turned off. But it is intermittent.

Your project is really the best. Many thanks.

pottedmeat7, 2020-10-23:
I will look into why it is crashing in that case. It might just be a combination of selected devices I haven't fully tested.
Thanks for the report, and I'm glad it worked out!

CFA44, 2020-10-26:
Thank you for saying so
I'll ask you one more question.
I tried touching the Enable WIP checkbox in desktop mode, but unlike the overlay state, the driver does not run. I tried using the administrator authority check, but there was no response. Did I miss something?
I want to test it in offline mode and in an environment without a controller.

CFA44, 2020-10-26:
I've done so many tests, and I have a question. The left hand controller must be turned on, but the OVRWIP plugin is visible on SteamVR.

[image]

OVRWIP does not work when only the right-hand controller is turned on, and it is not captured as a driver on SteamVR.
The manipulated value bound by the driver is seen as the value of the left hand controller. It seems that the left-handed controller must be paired to read the tracker's movement and apply it to the game. Would it be difficult even with both controllers turned off? I want to try to bind the manipulated values ​​in the game directly, but the test data is still insufficient.

pottedmeat7, 2020-10-27:
If your looking at the greyed out controller icon, it is the ovrwip_controller but it doesn't need to be lit up.
If it's not working with 1 controller, make sure have the correct number of controllers selected on the selection page. If it's configured for 2 controllers then it will wait for 2 controllers. 
I use it configured with only the right controller, so I can use the left controller for direction, while using the right hand for arm swinging.

CFA44, 2020-10-27:
If so, if both controllers are off, the driver is not recognized, and no input is supported even in in-game?
If both controllers are turned off while all are recognized, the operation will not work.
Is the WIP plugin an algorithm that works instead of the left controller?
I'm testing it in an environment where the controller is not supported, but it doesn't work.

In addition, if I assign a tracker role to Held in hand-left hand in SteamVR, the driver will be recognized but will not move.

pottedmeat7, 2020-10-29:
I'm not sure I fully understand your question. But from what I have tested as long as the device you have selected are turned on, and you have at least 1 device defined as a "controller" or "in hands" then it should work.
If the driver were on and powered off, that could cause issues.

CFA44, 2020-10-30:
Sorry. I am a little poor at English.

What I was talking about was the question of whether it is possible to run WIP with only a tracker in a situation where there is no controller of both hands at all.

Of course, the two controllers are not even paired.

pottedmeat7, 2020-10-30:
I will have to test that case and see if it works. I don't have my system together at the moment, it is in storage so won't be able to do anything for awhile.
One problem may be that most games expect some type of controller, left/right hand, also...