pottedmeat7/OpenVR-WalkInPlace #76: Setting input Keyboard(WASD) option problem

CFA44, 2020-10-29:
Hi 

I am currently testing with the keyboard (WASD) option.
However, in the input test, the W key was not pressed, so I checked that the shift key was pressed continuously.
Is there any way to fix the keyboard input (WASD)?
The keyboard (Arrow) option was also shifted down.

I always get a lot of help with this driver. Thanks for continuing to develop

pottedmeat7, 2020-10-29:
The W key said be ordered when you any value movement of detected. Check the data model and see if your getting a match.
It will only press the W it doesnt press shift also.

CFA44, 2020-10-30:
Yeah, so when I tested it with a keyboard tracker, the shift wasn't pressed. The indication that the W key is pressed consecutively appears, but it was not recognized in notepad or other games, and when the Keybaord (WASD) was activated, the capital letters appear as if shift was pressed when typing. 

I think I missed something, but I can't clearly see what it is. First of all, I was constantly testing, but I wondered if there was anything you knew about it.

CFA44, 2020-10-30:
[image]

Additional test material. When I input WASD, I tried using several testers, but the input value is not constant. Is it a problem with my computer settings?