pottedmeat7/OpenVR-WalkInPlace #77: Unable to enable

GKILLA1, 2021-01-02:
Hi potted, I followed the setup guide to the t but unfortunately I can't get it to work. I'm trying to track my vive trackers on feet to create data and use in vrchat. I recorded the data profile and set key bind in steam. After setup I saw the new device on steam hardware as a sort of hexagon with a C on it . when in steam vr home , its green stating active. as soon as I launch a game it greys out and goes into standby? is there anything you could suggest please ? I have attached a picture of the logs. 
[image]

GKILLA1, 2021-01-02:
also will this just use the games walking movement or will it simulator me walking ? when in vrchat using full body , the character walks using vr chats walk. will this just activate that walk when I WIP or will I walk like a human would ?

pottedmeat7, 2021-01-08:
If your 6 have the hexagon C, it's a good sign. Whether it's grey or not, doesn't seem to actually matter. 
The ovrwip_000 drive should still be able to take input.
Did you setup the controller bindings, and select to bind to left/right controller?

pottedmeat7, 2021-01-08:
Oops didn't mean to click close. 
You can test if the controller is getting input with the "test controller" option in steamvr, try testing on the ovrwio_000 device

dan9401, 2021-05-23:
Followed everything in the setup guide and binding guide. Having the same problem; however, seems a bit different for me. Sometimes is a blinking hexagon with c, sometimes is a grayed oculus touch controller, this message below is always there:
> This controller is not currently tracking. Make sure it's in view of a base station.

Tried to test the controller as suggested in steamVR, I have feedback when it shows the grayed out controller, but no feedback when it's the hexagon; never worked in-game(HLALYX). Any idea what the problem could be?

pottedmeat7, 2021-05-23:
Unfortunately it seems possible that some games can completely side step the controller bindings.
Yes in HLALYX, the controller binding system is broken. It requires that the analog stick be pushed, to start to enable movement, regardless of the bindings.
As an experiment try this:
Remove the movement input binding from your oculus controllers completely. 
Enable ovrwip and ensure you have input, then in game make sure the thumb stick doesn't move you by itself. Now begin walking in place, you won't move, until hold the thumb stick in any direction. There is no binding for the thumb stick but the game looks directly at the thumb stick ignoring bindings, so you will move only when it's pushed.
It really kind of defeats the purpose of having rebindable controller definitions when the game is programmed this way. 
Things should they alot better when OpenXR comes around. Also hopefully VR game developers will start to understand the VR paradigm a little better. Games won't have as many weird issues, and will definitely ensure support for 3rd controller systems, such as treadmills, which require some type of re-binding.
It's ironic that this issue is on HLALYX. Valve built support for the "/user/treadmill" controllers in OpenVR but HLALYX game developers weren't on the same page I guess.

pottedmeat7, 2021-05-23:
I  opened a ticket on OpenVR maybe they can do something... 
https://github.com/ValveSoftware/openvr/issues/1553
But obviously if the game developers cared they could fix this too...

dan9401, 2021-05-23:
Thanks. I tried to follow your advice today, just couldn't change the touch bindings.

Before starting the game, the bindings of the game are the touch legacy setting; after starting the game, it switches to "official bindings with offhand movement", and edits to the binding can't be saved (well not exactly, it did save once, god knows how, but mostly does nothing when you click save in save personal bindings).

I tried to manually delete the move section in the json file in User/Documents/steamvr/input, then everytime I start the game, the game will just revert the json file. Setting the file to read-only leads to a .tmp file created. I haven't tried setting read-only to the .tmp file yet, have a strong suspicion it will just crash or create another tmp.

pottedmeat7, 2021-05-23:
That's interesting, I was also having issues creating a saved binding at some point with HLALYX. 
After trying different approaches I was able to get walk in place to seemingly input into the game, but the game seems to require you hold the thumb stick which kind of defeats the purpose. 
If there was some trick with the input json definitions that would be excellent! Let us know if you have any luck!

pottedmeat7, 2021-05-27:
Ok got some great info from my ticket on OpenVR git. It is possible to get input in HLALYX just very unclear.
This is the binding you need to apply in the "Move" tab of ovrwip_controller binding. I also published the controller binding to the shared workshop, that should be available for ovrwip controllers. You also bind to left/right hand in "Extra settings" tab
[image]

dan9401, 2021-05-29:
Just got a chance to put on back my headset, I can confirm that this works. Now, recording a good data model ...