pottedmeat7/OpenVR-WalkInPlace #80: Tracked devices not saving

FlaggAC, 2021-05-23:
Pretty much the same as this issue, only none of the solutions proposed are helping. The INI file doesnt have an a entry for tracker IDs, just tracker indexes. 

https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/71

I am using valve index, two index controllers, two vive tracker 3.0 for feet. 

OpenVRWalkInPlace.ini
```
[walkInPlaceSettings]
walkInPlaceProfiles\1\profileName=a
walkInPlaceProfiles\1\modelFile=default
walkInPlaceProfiles\1\wipEnabled=false
walkInPlaceProfiles\1\gameType=0
walkInPlaceProfiles\1\hmdType=0
walkInPlaceProfiles\1\buttonControlSelect=0
walkInPlaceProfiles\1\useTrackers=true
walkInPlaceProfiles\1\trackHMDVel=true
walkInPlaceProfiles\1\trackHMDRot=true
walkInPlaceProfiles\1\disableButton=-1
walkInPlaceProfiles\1\minTouch=@Variant(\0\0\0\x87>\xb3\x33\x33)
walkInPlaceProfiles\1\midTouch=@Variant(\0\0\0\x87?333)
walkInPlaceProfiles\1\maxTouch=@Variant(\0\0\0\x87?\x80\0\0)
walkInPlaceProfiles\1\buttonEnables=0
walkInPlaceProfiles\1\cntrl1Idx=0
walkInPlaceProfiles\1\cntrl2Idx=1
walkInPlaceProfiles\1\trkr1Idx=0
walkInPlaceProfiles\1\trkr2Idx=-1
walkInPlaceProfiles\1\cntrl1Type=2
walkInPlaceProfiles\1\cntrl2Type=2
walkInPlaceProfiles\1\trkr1Type=3
walkInPlaceProfiles\1\trkr2Type=3
walkInPlaceProfiles\size=1

```

pottedmeat7, 2021-05-23:
The system will just use the 1st, 2nd, 3rd etc. tracker that is plugged in that is what is meant by the index.
Ie. If you had 2 trackers you would want the first, 0th index, and the second, 1st index
Regardless of the IDs, it would be 
```walkInPlaceProfiles\1\trkr1Idx=0
walkInPlaceProfiles\1\trkr2Idx=1
walkInPlaceProfiles\1\trkr1Type=3
walkInPlaceProfiles\1\trkr2Type=3```
Everything goes off of the order the devices are powered on. And type. The actual IDs will group controllers and trackers together, and the order is more confusing.

FlaggAC, 2021-05-23:
@pottedmeat7 okay I changed the INI file and restarted it and made a new data model. This is what I see (only HMD is part of data model):

[image]

[image]

[image]

I cannot verify whether this is solved because of the above graphs and because of https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/82 preventing anything from working

pottedmeat7, 2021-05-23:
Can you share a picture of the "tracked devices" page?

FlaggAC, 2021-05-23:
[image]

pottedmeat7, 2021-05-23:
That looks right. I'm not sure something different is happening in your case, I've never seen the issue with the test controllers in https://github.com/pottedmeat7/OpenVR-WalkInPlace/issues/82 either, so really unsure what's wrong. I would say probably try with 1 "default" profile, disable any extra devices maybe, or other drivers possibly...