pottedmeat7/OpenVR-WalkInPlace #86: no virtual controller is created

Nu70ry, 2023-10-28:
i created a data model enabled wip and choose my created data model and applied it.
but steamvr does not show an additional virtual controller.

pottedmeat7, 2023-10-28:
Check out which devices you used on the "Tracked Devices Page", I believe all of them need to be turned on before the virtual controller will be created. At least two controllers need to be powered on. If that doesn't work then maybe there is useful info in the driver or overlay logs...

Nu70ry, 2023-10-28:
hmd options both are enabled  controllers for pace only/compared to model both enabled controllers also vibrating if these get checked. other options cant be checked they uncheck themself after exiting the menu.
i dont know where i can find the driver/overlay logs.
thanks for your fast response

pottedmeat7, 2023-10-28:
Logs are here https://github.com/pottedmeat7/OpenVR-WalkInPlace#any-issues-check-out-the-logs

Nu70ry, 2023-10-28:
the driver log seems fine just "dll loadet" and the install path gets printed.
but the other log gets this

INFO ] 2023-10-29 00:48:07: unable to load data model at: C:/Users/battr/AppData/Roaming/pottedmeat7/OpenVRWalkInPlace/default.csv
[INFO ] 2023-10-29 00:48:07: Found device: id 0, class 1, serial 2G0YC1ZF7V0D27
[INFO ] 2023-10-29 00:48:07: Found device: id 1, class 2, serial 2G0YC1ZF7V0D27_Controller_Left
[INFO ] 2023-10-29 00:48:07: Found device: id 2, class 2, serial 2G0YC1ZF7V0D27_Controller_Right
[INFO ] 2023-10-29 00:48:40: Exception caught while adding initializing driver: Could not open server-side message queue: The system cannot find the file specified. driver_vrwalkinplace.server_queue

there is just 1.csv because i called my model "1"
if i rename it to default he finds the model and uses it.
but the exception still occurs and problem still persists.

pottedmeat7, 2023-10-28:
Hmm, so the driver isn't being fully loaded or is, and the client isn't able to connect.... 
Its possible the driver add on is being disabled by steamvr.
Check here 
[image]

to see, otherwise I'm not sure, it was working for me about a month or so ago, but maybe there was a steamvr update...

Nu70ry, 2023-10-29:
seems fine to me

[image]

Nu70ry, 2023-10-29:
steamvr 2.0 is out since a few days, dunno if its still compatible with it.

pottedmeat7, 2023-11-07:
Its possible that there are conflicts with the other add ons you have running. You might want to disable Natural Locomotion and 00vrocker. 
Could be one or both, but not entirely sure...