pottedmeat7/OpenVR-WalkInPlace #87: #USERTREADMILL shows input but games won't use, even with bindings set

EpochalEngineer, 2023-11-19:
I am using a Quest 3 and did the setup documented [here](https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/full_setup/Readme.md) and game-specific bindings [here](https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/steamvr_bindings/Readme.md). When I go to test controller under SteamVR it shows input under #USERTREADMILL when walking in place, but never actually moves the character. Is there more I need to setup other than the trackpad [here](https://github.com/pottedmeat7/OpenVR-WalkInPlace/raw/master/docs/screenshots/steamvr_bindings/steamvr_bindings_8.PNG) and [here](https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/steamvr_bindings/steamvr_bindings_10.PNG) ?

I tried it in a few games: After The Fall, Killing Floor, and GORN- none seem to work. Each one does say active controller is Oculus Touch with no option to change it, despite in edit controller bindings showing ovrwip_controller as current controller, not sure if that matters. The left control stick still works in all of these games for moving.

Also there is no option on my setup to override/emulate(?) my left or right controller directly, which I assume is this 2nd row dropdown shown [here](https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/docs/screenshots/full_setup/ovrwip_step5.PNG). The VRocker demo does seem to override/emulate the oculus touch left control stick, as when I have that turned on (and OVRWIP off), the real left control stick does not work, and the only way to walk is wakling in place.

What am I doing wrong?

SteamVR 2.1.4 but this also occurred when I was trying it out before SteamVR 2.0 came out.

EpochalEngineer, 2023-11-19:
Also, I seem to have a lot of inertia/momentum/lag, where it keeps moving for a full second or so after I stop. If there isn't a setting for this, where should I look in the code to change this?

pottedmeat7, 2023-11-20:
There is no option to directly emulate a physical controller. That's because most ways to implement that aren't really supported by the OpenVR SDK specification, and believe most of this project mostly is supported and should work without needed overriding.
Yes you may need to create a new binding for a game to get the input working. I had this problem with Half life Alyx, but there was binding that did allow it to work. 

One option for the controller relative direction that I now prefer to default to in my setup. I disable my non dominant hand that I use for direction in the model recording, usually your non dominant hand, because you may be aiming with your dominant hand. So to control pace, jogging etc., essentially it's just my right hand for instance, the left can stay complete still and just direct. 
Using the controller model recording is also a good way to improve the inertia, you may just prefer to disable the "walking" detection, and input 0, ie. to not move. So it may detect "walking" before it completely stops input, but either way your character will stop moving. 
But yeah it's not 100% synchronized timing to when you actually stopped, so there's always some lag.