pottedmeat7/OpenVR-WalkInPlace #91: Current state of Linux support?

lambdalemon, 2024-08-12:
Everything seems to compile (after some tweaks). Though I got a lot of warnings like `WARNING: Could not determine path of ... Skipping...` from linuxdeployqt, which seems kind of sus? (Is linuxdeployqt necessary? OVRAS also uses qt and seems to work without it on linux)

`~/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace` appears in the list of installed drivers in ALVR, but steamvr does not show any new device?

Also the back button and checkbox checks are not showing up (they are still functional). Here is the overlay log for reference.
[OpenVR-WalkInPlaceOverlay.log](https://github.com/user-attachments/files/16588128/OpenVR-WalkInPlaceOverlay.log)

Hopefully linux support is still planned?

pottedmeat7, 2024-08-12:
Ok glad to know it is at least compiling for others!
It was a bit if a struggle for a few things to get going on Linux with SteamVR for Linux etc. 
It has been a little while since I have tried again, I was kind of hoping maybe more recent OpenVR or SteamVR updates would also help get the Overlay and Driver working correctly. 
`linuxdeployqt` was an effort to provide all the necessary QT libs which are dependencies of the project included into the install path. However it's possible those dependencies are met in a global path, or other QT installation methods. Using that tool was just a similar approach to the windows install of this project.
If the overlay is loaded and reacting, showing data etc. That is a good sign the QT dependencies have been met! 
I remember I got about that far, the overlay side of things was running. Then there is the device driver side of things. 
Using boost IPC between the Overlay and Server. What I sort of remember is the WalkInPlace service was attempted to start and or successfully starts but then the IPC from the Overlay to Server wasn't never received... 
There is multiple logs provided by SteamVR. I forget the path but one log named `VRServer` in  the SteamVR installation somewhere. 
In that log you should see some useful details about SteamVR at least trying or successfully starting the WalkInPlace server. 
The WalkInPlace server also should be providing another log itself, both if those logs should be helpful, along with the Overlay log. 
Thought maybe it's a SteamvrForLinux process communication issue, I wasn't sure. Once the Server is able to receive messages from the Overlay, pretty much everything else should work. 
The third controller should be added and able to receive input etc. They just weren't talking before.

lambdalemon, 2024-08-13:
So I found this in vrserver.txt
```
Tue Aug 13 2024 21:55:06.017740 [Info] - Skipping duplicate external driver slimevr in "/home/lambdalemon/.steam/steam/steamapps/common/SteamVR/drivers/slimevr" because a driver of that name was already found at "/home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/slimevr"
Tue Aug 13 2024 21:55:06.017811 [Info] - Skipping duplicate external driver path "/home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace"
...
Tue Aug 13 2024 21:55:06.109039 [Info] - Unable to load driver 00vrwalkinplace from /home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace/bin/linux64/driver_00vrwalkinplace.so (/home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace/bin/linux64/driver_00vrwalkinplace.so: undefined symbol: _ZN2el4base5utils13s_currentHostB5cxx11E)
Tue Aug 13 2024 21:55:06.109051 [Error] - ASSERT: "DriverManager 00vrwalkinplace load error /home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace/bin/linux64/driver_00vrwalkinplace.so: undefined symbol: _ZN2el4base5utils13s_currentHostB5cxx11Eu" at /data/src/common/vrcommon/drivermanager.cpp:439.
Tue Aug 13 2024 21:55:06.130559 [Error] - Unable to load driver 00vrwalkinplace because of error VRInitError_Init_FileNotFound(103). Skipping.
```
It seems it was never successfully loaded. Also installed multiple times for some reason?

pottedmeat7, 2024-08-13:
Yes looks like a runtime dependency issue, it's possible easy logging support on Linux requires some changes compared to Windows usage. 
I forget exactly how easy logging is integrated, I thought it was a header library and compiled into the so, but maybe there's some other compile time linkage... It compiles, just doesn't find that function it's looking for. 
I do kind of remember having a similar issue, I think I tried completely removing easy logging, but I didn't get to replacing with any other type of logging. 
After removing I had some additional issues, with runtime dependencies, but at that point I had no logging which didn't help.

lambdalemon, 2024-08-14:
It seems easylogging++ used to be a header library but that got changed in a recent version.
Anyways I made some more changes and got the driver successfully loaded. The server log only contains a single line
```
[INFO ] 2024-08-15 00:24:56: Install Dir:/home/lambdalemon/.local/share/Steam/steamapps/common/SteamVR/drivers/00vrwalkinplace
```
IPC is still not working. Hopefully I'll figure this out tomorrow...

lambdalemon, 2024-08-15:
Actually it's working, I just didn't set the log level verbose enough. The check marks and back buttons are still missing though. Also the green line from show data model page isn't showing as well?

Another really weird thing is that somehow only the rotation of my HMD gets recorded. The position line seems to be a constant 0. Any idea what might be causing this? (Both track hmd and track hmd rotation are checked)

pottedmeat7, 2024-08-15:
Oh that's great! I'm hoping the other issues from the front end are mostly trivial 🤞
The check box etc not showing from what I remember was there was an issue with the drawing image support, there was something about that in the logs. Maybe a different QT dependency or newer image format would fix it...
For the HMD velocity values I would add some logs to the overlay and maybe just log 
Values of `latestDevicePoses[hmdID].vVelocity` after running `vr::VRSystem()->GetDeviceToAbsoluteTrackingPose(vr::TrackingUniverseStanding, 0.0f, latestDevicePoses, vr::k_unMaxTrackedDeviceCount);`
It should be able to get HMD velocity, my guess is something else is is breaking and not showing the correct values on the graph display. 
If it's not getting any velocity values, maybe something changed in the newer SteamVR or OpenVR libraries, 
I suppose if the checkboxes aren't showing then it would be hard to tell what's enabled or not, I remember it making some confusion...
But stick with it sounds like you are very close!

lambdalemon, 2024-08-16:
I figured out how to fix the back button (it was some peculiarity of linuxdeployqt.py)

As for the checkboxes, the only reference I've found is this [](https://doc.qt.io/qt-5/qquickimageprovider.html)
It doesn't mention a default image provider. Is `image://default/check/#ffffff` provided by Qt, or is it something written by you or some third party?

I was able to fix zero velocity by switching from "SteamVR" to "Other" (What does this option mean exactly?) Apparently some headsets don't report velocities and maybe quest 3 is one of them?

However, I am still unable to walk because `findMinPeakMN` never registered any peak, thus leading to invalid hmd sample. My headset Y velocity only reaches about 0.004 which is below the 0.007 required to be considered. The units for this seems to be m/ms, so 0.007m/ms = 7m/s. Isn't that insanely high? Or am I just bad at walking considering that it works for everyone else....

pottedmeat7, 2024-08-16:
Ah your using a Quest, yeah that is likely one of the ones that doesn't get direct velocity.
The "Other" option instead of using velocity, it gets the positions values directly from the pose matrix here is the two options https://github.com/pottedmeat7/OpenVR-WalkInPlace/blob/master/client_overlay/src/tabcontrollers/WalkInPlaceTabController.cpp#L466-L482
Then it calculates velocity from change in position. 
I'm really not sure what's coming out for values for that method, it does work when calculating the change of position for my headset an HTC Vive. But there maybe some change in the values as the direct velocity I think reports all 0s for those headsets too, so I'm not really sure why ..

lambdalemon, 2024-08-16:
Ok so I just checked the definition of `TrackedDevicePose_t` and turns out that all velocities are measured in seconds so the velocity should be scaled by 1000 when using "Other". It's quite surprising that "Other" actually works for you though...

Also let me know if you have any thoughts on the checkbox (see previous comment)?